armbian/build #1838: [new feature] NanoPi R1S H5 support

ryanm101, 2020-03-10:
Would like to add support for [NanoPi R1S](https://www.friendlyarm.com/index.php?route=product/product&path=69&product_id=274)

Basic functionality will work using the NanoPi Neo2 H5 Image, however it is missing eth1 and wlan0 devices (not sure what else isn't working yet).

Happy to help if someone can point me in the right direction to get started.

igorpecovnik, 2020-03-10:
Abridged version: create a new dts file (in the same location) based on FriendlyARM intial work:
https://github.com/friendlyarm/linux/blob/sunxi-4.14.y/arch/arm64/boot/dts/allwinner/sun50i-h5-nanopi-r1s.dts
and add this newly created DTS to Makefile and provide a patch. For kernel 5.4 (current) and 5.6 (dev), optional for 4.14 legacy. DTB is the only difference. Also peek into [R1S H3](https://github.com/armbian/build/commit/117f68ab9b86a25632d68b6fe1d0baea527c5274) work since this board is probably also using UART1 instead UART0 for serial console.

For more info, do some search here https://forum.armbian.com/forum/4-development/ and better to continue conversation there since this is not exactly an issue.

igorpecovnik, 2020-03-10:
Compared to Neo2 you probably only need to enable USB where second Ethernet is and enable wifi / BT bits. Since WiFi is probably the same, you better make use of (our adopted) dt bits from Neo2 plus2.

ryanm101, 2020-03-11:
Thanks will do, I wasnt sure where i add request for new HW but saw another request in this issue log.

igorpecovnik, 2020-03-11:
O.K.

> I wasnt sure where i add request for new HW but saw another request in this issue log.

When you have a working build target, open a [pull request](https://github.com/armbian/build/pulls) where we will comment what specifically needs to be fixed, added or changed. If anything.

In case you have troubles for preparing this merge request, use forum.

robsy, 2020-03-28:
Hi,

I have a R1S H5, and nanopineo2black image buster-current (dated 17.02.2020) is working quite well (both ethernet working, wifi working, 4 CPU working - but getting quite warm...). Kernel 5.4.20-sunxi64 (updated to 5.4.28-sunxi64) is doing a useable job.
So this should be a good starting-point.

Regards

Robert

jiushibushuo, 2020-08-16:
Aug 2020, any progress?

igorpecovnik, 2020-08-16:
> Aug 2020, any progress?

Still waiting for your contribution in brain power + time or cash to hire man power. Then the progress will start. Closing due to inactivity - reopen if someone shows interest.

closed 2020-08-16 by igorpecovnik