armbian/build #1843: ramlog journald settings

coelner, 2020-03-14:
I open the issue here as I request a change of the default journald settings. But it is related to zram and logrotate, therefore it could be moved to the armbian-config repo.

I'm not sure but this could be a bug in systemd-journald.

As stated here: [https://www.freedesktop.org/software/systemd/man/journald.conf.html](https://www.freedesktop.org/software/systemd/man/journald.conf.html)
 The default SystemMaxUse is 10% and SystemKeepFree 15% of the filesystem. SystemMaxFileSize is limited to 1/8 of SystemMaxUse. That should make the SystemMaxUse=5Mb and the SystemMaxFileSize=640kb big at a maximum count of 7 rotated files(Or did I miss something?)

```
root@cubietruck:/var/log.hdd/journal# mount | grep log
/dev/mmcblk0p1 on /var/folder2ram/var/log type ext4 (rw,noatime,nodiratime,discard,errors=remount-ro,commit=600)
folder2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,relatime)
folder2ram on /var/log.hdd type tmpfs (rw,nosuid,nodev,noexec,relatime)
/dev/zram0 on /var/log type ext4 (rw,relatime,discard)
root@cubietruck:/var/log.hdd/journal# df -h | grep log
/dev/zram0       49M   48M     0 100% /var/log
folder2ram     1005M   48M  958M   5% /var/log.hdd
./journal/158a2766af5340eeb736f1e0094800c9:
total 48M
-rw-r----- 1 root systemd-journal 8.0M Mar 14 05:42 system.journal
-rw-r----- 1 root systemd-journal 8.0M Mar 14 05:42 system@9463e6e41b124081973ec513aadd5513-0000000000000001-0005a09be21854a5.journal
-rw-r----- 1 root systemd-journal 620K Mar 14 05:42 system@9933e834e3b04256b0993698384af82f-0000000000000001-00059fe60b211a79.journal
-rw-r----- 1 root systemd-journal 8.0M Mar 14 05:42 system@9933e834e3b04256b0993698384af82f-000000000000eb42-0005a0468e1b090b.journal
-rw-r----- 1 root systemd-journal    0 Mar 14 05:42 user-1000.journal
-rw-r--r-- 1 root systemd-journal  12K Mar 14 05:42 user-1000@0005a0c917562a56-03264fd7fbb9cab4.journal~
-rw-r----- 1 root systemd-journal 8.0M Mar 14 05:42 user-1000@48cb3359bebc48c5a365cd8331f44449-00000000000005a6-00059ff5cc5dc955.journal
-rw-r--r-- 1 root systemd-journal    0 Mar 14 05:42 user-1001.journal
-rw-r--r-- 1 root systemd-journal 6.6M Mar 14 05:42 user-1001@0005a04bd279d5f2-7db7110036687b2e.journal~
-rw-r----- 1 root systemd-journal 8.0M Mar 14 05:42 user-1001@e39d4f761e494905b53fc5c6f90f23e7-000000000000d12d-0005a03d450b6ec8.journal
root@cubietruck:/var/log# journalctl --vacuum-time=2d
Journal file /var/log/journal/158a2766af5340eeb736f1e0094800c9/user-1000.journal is truncated, ignoring file.
Vacuuming done, freed 0B of archived journals from /var/log/journal.
Deleted empty archived journal /var/log/journal/158a2766af5340eeb736f1e0094800c9/user-1000@0005a0c917562a56-03264fd7fbb9cab4.journal~ (16.0K).
Deleted archived journal /var/log/journal/158a2766af5340eeb736f1e0094800c9/system@9933e834e3b04256b0993698384af82f-0000000000000001-00059fe60b211a79.journal (620.0K).
Deleted archived journal /var/log/journal/158a2766af5340eeb736f1e0094800c9/user-1000@48cb3359bebc48c5a365cd8331f44449-00000000000005a6-00059ff5cc5dc955.journal (8.0M).
Deleted archived journal /var/log/journal/158a2766af5340eeb736f1e0094800c9/user-1001@e39d4f761e494905b53fc5c6f90f23e7-000000000000d12d-0005a03d450b6ec8.journal (8.0M).
Deleted archived journal /var/log/journal/158a2766af5340eeb736f1e0094800c9/system@9933e834e3b04256b0993698384af82f-000000000000eb42-0005a0468e1b090b.journal (7.9M).
Deleted archived journal /var/log/journal/158a2766af5340eeb736f1e0094800c9/user-1001@0005a04bd279d5f2-7db7110036687b2e.journal~ (6.5M).
Deleted archived journal /var/log/journal/158a2766af5340eeb736f1e0094800c9/system@9463e6e41b124081973ec513aadd5513-0000000000000001-0005a09be21854a5.journal (8.0M).
Vacuuming done, freed 39.2M of archived journals from /var/log/journal/158a2766af5340eeb736f1e0094800c9.
```
My questions:
1. is this in the right issue tracker?
2. any thoughts about a workaround? systemd timer for vacuuming journal logs?

igorpecovnik, 2020-03-14:
> But it is related to zram

ZRAM is installed / enabled by default 

> is this in the right issue tracker?

which means we have to fix this here.

Which OS version do you use? Or just provide logs with: `armbianmonitor -u`

coelner, 2020-03-14:
Armbian buster Linux 5.4.20-sunxi on a cubietruck
armbianmonitor -u : http://ix.io/2ece

igorpecovnik, 2020-03-19:
Perhaps setting maximum value would do?

`sudo sed -i "s/.*SystemMaxUse=$/SystemMaxUse=10M/" /etc/systemd/journald.conf`

coelner, 2020-03-21:
I'm going to try this.

coelner, 2020-03-25:
```
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:30 system@d9348c551249486da2bc2c75cb2f59a0-0000000000579970-0005a1addebcb0b8.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 system@d9348c551249486da2bc2c75cb2f59a0-0000000000579ca1-0005a1addfd669c6.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 system@d9348c551249486da2bc2c75cb2f59a0-0000000000579fee-0005a1ade68c5fc7.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 system@d9348c551249486da2bc2c75cb2f59a0-000000000057a33c-0005a1ade6be76d2.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 system@d9348c551249486da2bc2c75cb2f59a0-000000000057a690-0005a1ade6e1cafe.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 system@d9348c551249486da2bc2c75cb2f59a0-000000000057a9d8-0005a1ade7054302.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 system@d9348c551249486da2bc2c75cb2f59a0-000000000057ad25-0005a1ade72cc039.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:34 system.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 user-1001.journal
-rw-r-----+ 1 root systemd-journal 1,3M Mär 25 14:32 user-1002.journal
```

it seems to work

closed 2020-03-25 by coelner