armbian/build #1847: can not run docker on Fedora 31

gahabana, 2020-03-15:
hi, i cloned repository on Fedora 31 where i have docker running just fine. 
on the 1st run:
`./compile docker`
i get message that repo.aptly.info has invalid signature.

I hope there is a solution for non-ubuntu users ?
`
zh@acus:~/git/armbian-build$ sudo ./compile.sh docker
[ o.k. ] This script will try to update
Already up to date.
Already on 'master'
Your branch is up to date with 'origin/master'.
which: no dialog in (/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin)
sudo: apt: command not found
sudo: apt: command not found
[ o.k. ] Create example config file using template [ config-default.conf ]
[ o.k. ] Using config file [ /home/zh/git/armbian-build/userpatches/config-docker.conf ]
[ .... ] Docker container not found or out of date
[ .... ] Building a Docker container
Sending build context to Docker daemon  16.38kB
Step 1/10 : FROM ubuntu:18.04
 ---> 72300a873c2c
Step 2/10 : RUN apt update && DEBIAN_FRONTEND=noninteractive apt install -y gnupg gnupg1 gpgv1 --no-install-recommends
 ---> Using cache
 ---> 7f5777d6b838
Step 3/10 : RUN sh -c  "if [ x"" != x$http_proxy ]; then                 apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --keyserver-options http-proxy=$http_proxy --recv-keys ED75B5A4483DA07C >/dev/null 2>&1;  		    else 			    apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys ED75B5A4483DA07C >/dev/null 2>&1; 		    fi"
 ---> Using cache
 ---> e8dbb94fdc5d
Step 4/10 : RUN echo "deb http://repo.aptly.info/ nightly main" > /etc/apt/sources.list.d/aptly.list
 ---> Using cache
 ---> 0e9b15115947
Step 5/10 : RUN dpkg --add-architecture i386
 ---> Using cache
 ---> 5febf5db445c
Step 6/10 : RUN apt update && DEBIAN_FRONTEND=noninteractive apt -y upgrade && 	DEBIAN_FRONTEND=noninteractive apt install -y git dialog lsb-release binutils wget ca-certificates device-tree-compiler 	pv bc lzop zip binfmt-support build-essential ccache debootstrap ntpdate gawk gcc-arm-linux-gnueabihf 	qemu-user-static u-boot-tools uuid-dev zlib1g-dev unzip libusb-1.0-0-dev parted pkg-config libncurses5-dev whiptail 	debian-keyring debian-archive-keyring f2fs-tools libfile-fcntllock-perl rsync libssl-dev nfs-kernel-server btrfs-progs 	ncurses-term p7zip-full kmod dosfstools libc6-dev-armhf-cross fakeroot xxd 	curl patchutils python liblz4-tool libpython2.7-dev linux-base swig libpython-dev python3-dev 	systemd-container udev g++-5-arm-linux-gnueabihf lib32stdc++6 cpio tzdata psmisc acl 	libc6-i386 lib32ncurses5 lib32tinfo5 locales ncurses-base zlib1g:i386 pixz bison libbison-dev flex libfl-dev 	pigz aptly aria2 cryptsetup cryptsetup-bin python3-distutils --no-install-recommends
 ---> Running in c2f02aec48b4

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Hit:1 http://security.ubuntu.com/ubuntu bionic-security InRelease
Hit:2 http://archive.ubuntu.com/ubuntu bionic InRelease
Hit:3 http://archive.ubuntu.com/ubuntu bionic-updates InRelease
Hit:4 http://archive.ubuntu.com/ubuntu bionic-backports InRelease
Get:5 http://security.ubuntu.com/ubuntu bionic-security/restricted i386 Packages [5385 B]
Get:6 http://security.ubuntu.com/ubuntu bionic-security/main i386 Packages [567 kB]
Get:7 http://archive.ubuntu.com/ubuntu bionic/main i386 Packages [1328 kB]
Get:8 http://repo.aptly.info nightly InRelease [7495 B]
Get:9 http://security.ubuntu.com/ubuntu bionic-security/multiverse i386 Packages [4513 B]
Get:10 http://security.ubuntu.com/ubuntu bionic-security/universe i386 Packages [785 kB]
Err:8 http://repo.aptly.info nightly InRelease
  The following signatures were invalid: EXPKEYSIG ED75B5A4483DA07C Andrey Smirnov <me@smira.ru>
Get:11 http://archive.ubuntu.com/ubuntu bionic/restricted i386 Packages [13.5 kB]
Get:12 http://archive.ubuntu.com/ubuntu bionic/multiverse i386 Packages [177 kB]
Get:13 http://archive.ubuntu.com/ubuntu bionic/universe i386 Packages [11.3 MB]
Get:14 http://archive.ubuntu.com/ubuntu bionic-updates/restricted i386 Packages [14.1 kB]
Get:15 http://archive.ubuntu.com/ubuntu bionic-updates/universe i386 Packages [1296 kB]
Get:16 http://archive.ubuntu.com/ubuntu bionic-updates/main i386 Packages [843 kB]
Get:17 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse i386 Packages [8174 B]
Get:18 http://archive.ubuntu.com/ubuntu bionic-backports/main i386 Packages [2493 B]
Get:19 http://archive.ubuntu.com/ubuntu bionic-backports/universe i386 Packages [4240 B]
Reading package lists...
W: GPG error: http://repo.aptly.info nightly InRelease: The following signatures were invalid: EXPKEYSIG ED75B5A4483DA07C Andrey Smirnov <me@smira.ru>
E: The repository 'http://repo.aptly.info nightly InRelease' is not signed.
The command '/bin/sh -c apt update && DEBIAN_FRONTEND=noninteractive apt -y upgrade && 	DEBIAN_FRONTEND=noninteractive apt install -y git dialog lsb-release binutils wget ca-certificates device-tree-compiler 	pv bc lzop zip binfmt-support build-essential ccache debootstrap ntpdate gawk gcc-arm-linux-gnueabihf 	qemu-user-static u-boot-tools uuid-dev zlib1g-dev unzip libusb-1.0-0-dev parted pkg-config libncurses5-dev whiptail    	debian-keyring debian-archive-keyring f2fs-tools libfile-fcntllock-perl rsync libssl-dev nfs-kernel-server btrfs-progs 	ncurses-term p7zip-full kmod dosfstools libc6-dev-armhf-cross fakeroot xxd 	curl patchutils python liblz4-tool libpython2.7-dev linux-base swig libpython-dev python3-dev 	systemd-container udev g++-5-arm-linux-gnueabihf lib32stdc++6 cpio tzdata psmisc acl 	libc6-i386 lib32ncurses5 lib32tinfo5 locales ncurses-base zlib1g:i386 pixz bison libbison-dev flex libfl-dev 	pigz aptly aria2 cryptsetup cryptsetup-bin python3-distutils --no-install-recommends' returned a non-zero code: 100

[ error ] Docker container build exited with code:  [ 0 ]
zh@acus:~/git/armbian-build$`

Thank you !

igorpecovnik, 2020-03-15:
ATM I can only confirm that its indeed a bug, but sadly not in our domain. I also doesn't work on Debian based distribution.

I will try to find a workaround ASAP.

zhangn1985, 2020-03-16:
aptly repo sign key is invalid. please wait aptly to fix it.

igorpecovnik, 2020-03-16:
Yes, issue already open:
https://github.com/aptly-dev/aptly/issues/911

igorpecovnik, 2020-03-17:
@gahabana You need to start from a clean fork or adjust userpatches/Dockerfile manually.

closed 2020-03-17 by igorpecovnik