armbian/build #1849: Build legacy 3.4 for nanopiduo2

roderick1980, 2020-03-18:
Hello,

I've been trying myself to build an image for nanopiduo, in order to get something similar to the "nanopi-duo2_sd_friendlycore-xenial_3.4_armhf_20181218.img".
However, it seems that compile.sh only allows me to create images from 4.x.

I'd like to create a 3.4 image for nanopiduo2 and also to add some customization for HW accelerated GPU video decoding.

I'd appreciate any help with this.

Thanks and kind regards

igorpecovnik, 2020-03-18:
> HW accelerated GPU video decoding.

Is enabled in 5.4.y as experimental feature
https://github.com/armbian/build/blob/master/config/kernel/linux-sunxi-current.config#L5596-L5598

> I'd like to create a 3.4 image for nanopiduo2 

Checkout to v19.08 and use LIB_TAG="v19.08" when building. In case of troubles, you are on your own. We can't afford to support that.

closed 2020-03-18 by igorpecovnik

roderick1980, 2020-03-18:
Thanks for your quick response.
Just one more question: does 5.4.y allow TV out (ananlog CVBS out)?
Kind regards

igorpecovnik, 2020-03-18:
> does 5.4.y allow TV out (ananlog CVBS out)?

I don't know for sure how far this function is or if its working at all. Do some forum searching https://forum.armbian.com/