armbian/build #1859: Patch Ext WiFi

150balbes, 2020-03-29:
Fixed the build of WiFi modules (Ext) for RTL on kernel 5.6, tried to add patches to the build system, but for a reason that I don't understand, the patches are not applied. I did it in a stupid way - I built a kernel branch with added fixed sources. Perhaps someone will understand the reason why patches are not included. Example of adding a module for rtl8189es and a patch. Patches (diff) can be taken from a branch with fixed sources.

`	# Wireless drivers for Realtek 8189ES chipsets

	if linux-version compare $version ge 3.14 && [ "$EXTRAWIFI" == yes ]; then

		# attach to specifics tag or branch
		local rtl8189esver="branch:master"

		display_alert "Adding" "Wireless drivers for Realtek 8189ES chipsets ${rtl8189esver}" "info"

		fetch_from_repo "https://github.com/jwrdegoede/rtl8189ES_linux" "rtl8189es" "${rtl8189esver}" "yes"
		cd ${SRC}/cache/sources/${LINUXSOURCEDIR}
		rm -rf ${SRC}/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/rtl8189es
		mkdir -p ${SRC}/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/rtl8189es/
		cp -R ${SRC}/cache/sources/rtl8189es/${rtl8189esver#*:}/{core,hal,include,os_dep,platform} \
		${SRC}/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/rtl8189es

		# Makefile
		cp ${SRC}/cache/sources/rtl8189es/${rtl8189esver#*:}/Makefile \
		${SRC}/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/rtl8189es/Makefile
		cp ${SRC}/cache/sources/rtl8189es/${rtl8189esver#*:}/Kconfig \
		${SRC}/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/rtl8189es/Kconfig

		# Patch
		cd ${SRC}/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/rtl8189es/
		process_patch_file "${SRC}/patch/misc/wireless-rtl8189es.patch"                "applying"
		cd ${SRC}/cache/sources/${LINUXSOURCEDIR}

		# Add to section Makefile
		echo "obj-\$(CONFIG_RTL8189ES) += rtl8189es/" >> $SRC/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/Makefile
		sed -i '/source "drivers\/net\/wireless\/ti\/Kconfig"/a source "drivers\/net\/wireless\/rtl8189es\/Kconfig"' \
		$SRC/cache/sources/${LINUXSOURCEDIR}/drivers/net/wireless/Kconfig

	fi
`

Patch

`--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -348,10 +348,9 @@
 
 static u64 rtw_get_systime_us(void)
 {
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,20,0))
-	struct timespec ts = ktime_to_timespec(ktime_get_boottime());
-	return ((u64)ts.tv_sec*1000000) + ts.tv_nsec / 1000;
-#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,39))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 20, 0))
+	return ktime_to_us(ktime_get_boottime());
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 39))
 	struct timespec ts;
 	get_monotonic_boottime(&ts);
 	return ((u64)ts.tv_sec*1000000) + ts.tv_nsec / 1000;`

150balbes, 2020-03-29:
I have patches for 8188 8723 and 88x2, do you need them ?

igorpecovnik, 2020-03-29:
>  have patches for 8188 8723 and 88x2, do you need them ?

Patches to run at 5.6 ?

150balbes, 2020-03-29:
> > have patches for 8188 8723 and 88x2, do you need them ?
> 
> Patches to run at 5.6 ?

yes

igorpecovnik, 2020-03-29:
yes, make a PR. We will need them sooner or later anyway.
 tnx

closed 2020-03-29 by igorpecovnik

150balbes, 2020-03-29:
ok