armbian/build #1878: build repository url incorrect in /etc/armbian-release

count-doku, 2020-04-09:
Hi,

the build_repository_url in /etc/armbian-release is always set to 'https://github.com/armbian/rkbin'
This url seems to be set here: https://github.com/armbian/build/blob/113f574e7086f8a7cffd2146ecea224eb1bd5bb1/lib/configuration.sh#L25

Example:
```
root@clearfogpro:~# cat /etc/armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=clearfogpro
BOARD_NAME="Clearfog Pro"
BOARDFAMILY=mvebu
BUILD_REPOSITORY_URL=https://github.com/armbian/rkbin
BUILD_REPOSITORY_COMMIT=3e533c9
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=20.02.1
LINUXFAMILY=mvebu
BRANCH=current
ARCH=arm
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=Image
```

Greetings,
count-doku

count-doku, 2020-04-11:
Interestingly with self build images the BUILD_REPOSITORY_URL is correct. So this seems like an issue on the build system only.

igorpecovnik, 2020-04-11:
Yeah. This build_all.sh part only we use and its not in a very good shape :( But this info helps at least to narrow the problem down a bit.

closed 2020-04-30 by igorpecovnik