armbian/build #1882: Add image for many Amlogic, Allwinner, and Rockchip devices

probonopd, 2020-04-11:
https://github.com/150balbes/Build-Armbian has a "universal ARM 64 image" configuration that allows different Amlogic, Allwinner, and Rockchip devices to boot using the mainline kernel, all from the same image. I have successfully tested this on an Amlogic-based x96 2GB box (which is currently not supported yet by Armbian proper).

I would like to see support for this in the official Armbian so that official Armbian images get built, not just in https://github.com/150balbes/Build-Armbian.

I don't quite know how to send a pull request that would enable this, but I figure that it should be something along those lines:

* https://github.com/150balbes/Build-Armbian/blob/master/config/boards/arm-64.conf to https://github.com/armbian/build/tree/master/config/boards
* https://github.com/150balbes/Build-Armbian/tree/master/packages/bsp/arm-64 to https://github.com/armbian/build/tree/master/packages/bsp
* https://github.com/150balbes/Build-Armbian/tree/master/patch/kernel/arm-64-current to https://github.com/armbian/build/tree/master/patch/kernel

paolosabatino, 2020-04-11:
That amlogic tvbox is not supported by "Armbian proper" and probably never will. Forget about "universal image", this is incompatible with "Armbian proper".

Ask the fork's author to make a **proper** pull request on his/her issue tracker if you want support.

probonopd, 2020-04-12:
Thanks @paolosabatino I respect your knowlededge but could you please elaborate a bit what you mean by "not supported by Armbian proper and probably never will".

As you might be aware, there has been a huge effort ongoing to bring the mainline kernel to Amlogic devices, and I am happy to say that the "universal image" I mentioned above I can successfully run the mainline kernel on said box.

The reason I am asking here is that I would like to see more Amlogic (and possibly also Allwinner, Rockchip) boxes using the mainline kernel appear on https://www.armbian.com/download/.

Can you describe what would need to be done to make this happen, and what specifically you think is incompatible? I'd be happy to support the effort if I knew where to start and what would need to be done to do this "properly".

igorpecovnik, 2020-04-12:
> As you might be aware

As you might not be aware, that project was supported with a bag of cash. Nobody there worked for free, while Armbian is an amateur project which receives around 3.000 EUR per year from you. That is not sufficient to cover costs users make by downloading binaries or a costs you make in a single day. This is financial part.

Next. Why investing huge amount of work, time and cash to make more depths? We don't gain in any way. Only you or lets say public gain. Haven't we already done enough?

Shell we close this "issue" now? We collect those ideas here: https://forum.armbian.com/forum/38-feature-requests/ where they wait for financial backing. If there is none, there is nothing we can do about. We are not a bank.

paolosabatino, 2020-04-12:
I want to highlight I am not a developer from Armbian project. I occasionally contribute, but I am not directly involved so, whatever I say, it is personal opinion and does not necessarily reflect the project guideline.
This issue tracker probably is not the right place to talk about. Tvboxes are not officially supported by Armbian statement, but they are brought on by community efforts, so the place to talk about is in the community forums: [Tv boxes sections](https://forum.armbian.com/forum/24-tv-boxes/) [Recent discussion on tv boxes issues](https://forum.armbian.com/topic/13611-infrastructure-needed-for-releasing-new-kernels-for-armbian-for-tv-boxes/)

probonopd, 2020-04-12:
Sorry @igorpecovnik if my question offended you in any way, that was entirely not my intention. I am very grateful that this project exists and for the hard work you are doing.

I understand that this is an unfunded volunteer project. I am working on such a project myself. Wouldn't your argument, though, apply to basically _any_ kind of feature request? What if the requestor offered to do the work?

@paolosabatino 

> Tvboxes are not officially supported by Armbian statement

Let's not talk about any particular "boxes", lets talk about "chipsets" then.

igorpecovnik, 2020-04-12:
> if my question offended you in any way

Not, but further will.

aadityabagga, 2020-04-13:
Hi @probonopd 

I have been following the Armbian project for a few months now. As far as I understand, Armbian does not want to add support for these boxes because there is lack of support from upstream board manufacturers.

Besides adding software support for the boards as you said, users come to the Armbian project for technical support, feature requests, etc. For this resources are needed.

Thanks for your work, you can keep contributing to Balbes' fork I guess. :slightly_smiling_face:

probonopd, 2020-04-13:
Thanks @aadityabagga, guess I have to keep downloading from "random" Russian websites at seemingly ever-changing URLs then. Sigh.

igorpecovnik, 2020-04-13:
> keep downloading from "random" Russian websites

I think you should be thankful. You contribute nothing and you have something much better than nothing to download. I know its not perfect, it also throws a bad light on our project and opens questions as this one. Once you want top speed verified download with top quality support, things become expensive, very expensive. You need to have a business backing - we generate no income. You don't even pay us for hours we waste for you addressing to us (dealing with this issue cost me let's say 200 EUR - I don't need your refund, just saying that things we don't want to do but you make us, are not going to be done for free)

Help doing something we don't plan nor want to do is not going to change things. We still need to guide you (on our costs we can't) how system works and pay for rest (95% which again we can't) of your idea ... top level changes are not done by opening an issue.

Few days back I accidentally bump into this tweet:
https://twitter.com/neil_neilzone/status/1248930281660301313

> Hello open source project, you don’t have Feature X!!!!!!!!!!
> 
> > Correct, it doesn’t fit with what we’re doing. You could pay someone to make the mods you need?
> 
> Pay? This is *Free* software!!!!!!
> 
> > yes, but we’re not your dev team
> 
> But why should I pay?
> 
> > aaaaargh

But here is not just about the feature. When we waste a million we don't want to get your where you want, where to find another million (per year) to cover support costs for all those devices? We already don't want to do that for free. 1 toy is fun, two perhaps still, but more are already a job. And nobody wants that this becomes their unpaid full-time job. Paid, perhaps and there are very small % of those.

The only way we can keep this project on quality levels (which you must have noticed and which is why you want this) is by limiting the hw we support. Those boxes makes costs to us - TV box is estimated on 10.000 EUR per year, for boards this varies between 20 and 50.000 EUR (except those which are re-branded version of another). Dealing with hardware and with users on the other hand represent a cost, a stress we are - in such extend - only willing to trade for money. If not, sources are out here ... do it yourself. If you know how, if you have patience and time.

I am sure you have good intentions and I also hope you understand our side better.

closed 2020-04-13 by igorpecovnik