armbian/build #1891: Enable ifb module with sunxi64

idubnori, 2020-04-16:
For the same reason as #568, I would like to enable `ifb` and use the input bandwidth limiting.

I tried the followings, but no luck...
Is the configuration insufficient or not supported?
1. Copy [linux-sunxi64-current.config](https://github.com/armbian/build/blob/master/config/kernel/linux-sunxi64-current.config) to`./userpatches/`and add configurations
```bash
CONFIG_NET_SCH_INGRESS = m
CONFIG_IFB = m
```

2. Build
```bash
./compile.sh BOARD=nanopineo2 BRANCH=current RELEASE=buster BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_KEEP_CONFIG=no
```
3. Replace kernel
Follow [the same steps] (https://forum.armbian.com/topic/12170-replace-the-kernel-to-the-custom-compiled-kernel/?ct=1587049950)

igorpecovnik, 2020-04-16:
We are working to come up with a better config, check https://github.com/armbian/build/pull/1856 and create a merge request to that branch. Not sure when we are going to merge that it, but soon.

idubnori, 2020-04-16:
Thank you for the quick!

Does it means currently we can not use that module using `.config` file with sunxi64 on current branch?

idubnori, 2020-04-18:
Thank you for your reply.

I've confirmed the followings using NanoPi NEO2.  Just information.
`linux-sunxi64-current.config` of the current #1856 might have an issue. 

- Failed to boot using  the image or kernel built with #1856
  - Can't get the boot logs and so on, due to I don't have serial connection device. Sorry, couldn't cooperation.

- With main branch with using #1876 and `CONFIG_IFB = m` only, could be boot and inbound traffic control using `ifb`

igorpecovnik, 2020-04-18:
Thank you for testing. That MR was only tested for building at this point and its possible that it fails. Its still some work to get where we want.

idubnori, 2020-04-18:
Got it. Thank you for the explanation.  Looking forward to be merged `AR-182` into the next release.

igorpecovnik, 2020-04-28:
https://github.com/armbian/build/pull/1856 merged

closed 2020-04-28 by igorpecovnik