armbian/build #1910: rootfs image for dedian should use debian's debootstrap

zhangn1985, 2020-04-23:
ubuntu's debootstrap, at least 18.04's debootstrap, may be it's too old, or missync with debian. Buster's rootfs image is created different to it created by buster's debootstrap script.

while this not lead any issue, just a notice.

igorpecovnik, 2020-04-23:
Do you think it is critical in any way to deboostraping Bullseye with Debootsrap from 18.04? We actually talked once about having our own deboostrap package, but I am not sure if this is a simple packagain install or not. On the other hand, we would not want to change host machine more than necessary. Or making complex installations. 

Building natively on 20.04 is already supported which should solve related (possible) issues, right?

igorpecovnik, 2020-05-08:
Linux mint Debbie:

```
[ o.k. ] Creating build chroot [ focal/arm64 ]
E: No such script: /usr/share/debootstrap/scripts/focal
[ error ] ERROR in function create_chroot [ chroot-buildpackages.sh:53 ]

```

So I guess we could do that ...

zhangn1985, 2020-05-09:
I guess to use debian's debootstrap to build debian, and ubuntu's debootstrap to build ubuntu.

igorpecovnik, 2020-05-09:
... and we don't want to mess up with the host system, which means we should have both scripts and support scripts installed elsewhere?

iav, 2022-04-02:
obsoleted?

igorpecovnik, 2022-04-02:
CI almost entirely depends on docker images and with generating different Docker bases  

https://github.com/armbian/scripts/blob/master/.github/workflows/update-docker-image.yml#L15
https://github.com/armbian/build/pkgs/container/build

we are getting closer to cover this idea. But I found one anomaly already (perhaps i didn't look properly) - with stock version of deboostrap on Bullseye it was not possible to make bullseye variant ...

This has to be well tested before proceeding ...

igorpecovnik, 2022-11-25:
A system for compiling kernels and headers with their distro compilers is now in the POC state. https://github.com/armbian/build/pull/4487 I'll close this issue for now as its not really an issue. After this is implemented we might also think to switch deboostrap to this logic. Cache is generate inside containers anyway, so we just need to choose right one ...

closed 2022-11-25 by igorpecovnik