armbian/build #1948: Build options LIB_TAG is ignored

g-provost, 2020-05-07:
I tried to build an image based on a branch (e.g AR-159) and I just realized that LIB_TAG param is not taken in consideration anymore. So it seems I have to merge first into master localy.

I haven't looked into the build script yet to find the reason. 

Wondering if it is a known issue. I remember the build script use to check out by default into master branch and do a git pull.

igorpecovnik, 2020-05-07:
Even you put to the config?

g-provost, 2020-05-08:
Actually the 1st parameter after docker|vagrant is ignored whichever is the param. In my case it was LIB_TAG, then I started to play around and realized that it was an issue on how the cmdline parametes are parsed.

g-provost, 2020-05-08:
It seems LIB_TAG is still ignore anyhow.

This time I'm triggering the build from within the VM to avoid the above issue.

```
vagrant@ubuntu-bionic:~/armbian$ sudo ./compile.sh BOARD="helios4" LIB_TAG="AR-159" BRANCH="current" RELEASE="buster" KERNEL_ONLY="no" KERNEL_CONFIGURE="no" INSTALL_HEADERS="yes" BUILD_DESKTOP="no" BUILD_MINIMAL="yes"
[ o.k. ] Using config file [ /vagrant/userpatches/config-example.conf ]
[ o.k. ] Command line: setting BOARD to [ helios4 ]
[ o.k. ] Command line: setting LIB_TAG to [ AR-159 ]
[ o.k. ] Command line: setting BRANCH to [ current ]
[ o.k. ] Command line: setting RELEASE to [ buster ]
[ o.k. ] Command line: setting KERNEL_ONLY to [ no ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ no ]
[ o.k. ] Command line: setting INSTALL_HEADERS to [ yes ]
[ o.k. ] Command line: setting BUILD_DESKTOP to [ no ]
[ o.k. ] Command line: setting BUILD_MINIMAL to [ yes ]
```

So the LIB_TAG is well paced, however the resulting image is not from AR-159 and nowhere in the build log i see the script doing a git checkout to the branch AR-159.

So in total 2 issue
* 1st Build param ignored when using docker | vagrant option
* LIB_TAG is ignored

The-going, 2020-06-16:
> I remember the build script use to check out by default into master branch and do a git pull.

This branch update mechanism does not currently work in the VM.
I fixed it here. https://github.com/armbian/build/pull/2023
But in any case, the user will have to do manipulations with the current branch independently.

If you implement simple branch switching, you will need to add the following actions:
1 Check the status of the current branch.
2 Check the existence of the branch \ tag in the remote repository.
3 Process the situation for errors.
4 Create a new branch and switch to it.
If something went wrong, stop and tell the user "Do it yourself"
A few more steps with manipulations in the git will lead to the fact that the action analyzer will grow a long beard, and the result will be a message to the user "Do everything yourself".

closed 2023-04-29 by rpardini