armbian/build #1961: Error when following vagrant setup guide

wilkgr76, 2020-05-11:
Hi all!

I'm following the [instructions for building with Vagrant](https://docs.armbian.com/Developer-Guide_Using-Vagrant/). I already had Vagrant installed so I liked seeing it!

When I get to the `sudo ./compile.sh` step, it appears to install dependencies from apt and then ln complains about a protocol error:
```
$ sudo ./compile.sh
[... apt output ...]
[ o.k. ] Create example config file using template [ config-default.conf ]
ln: failed to create symbolic link '/home/vagrant/armbian/userpatches/config-default.conf': Protocol error
```
I decided to just try running the command again, and now realpath complains about an empty parameter, which I'm guessing is because the symlink wasn't made:

```
$ sudo ./compile.sh
[ o.k. ] Create example config file using template [ config-default.conf ]
realpath: '': No such file or directory
[ .... ] Config file does not exist
```

igorpecovnik, 2020-05-20:
I do not use Vagrant / never did, so there is little I can help. This feature was brought up by community ...

@tianyuanhao Any idea why this would be happening?

tianyuanhao, 2020-05-21:
@wilkgr76 I guess you use Vagrant on Windows. If so, it should be because NTFS does not support symlinks well. In fact, NTFS does support symlinks, but requires some complicated settings. You can try to comment out `ln -sf /vagrant/userpatches /home/vagrant/armbian/userpatches` in `config/templates/Vagrantfile` and try again.

By the way, [this document](https://docs.armbian.com/Developer-Guide_Using-Vagrant/) is outdated, you can directly use the command `./compile.sh vagrant`.

wilkgr76, 2020-05-21:
Ahh thank you. Yes, I'm using Windows. I'll redownload and check if that works later tonight. Thank you!

tianyuanhao, 2020-05-21:
The safest way is to clone this project to a separate location in the box and make sure not to create any symlink to `/vagrant`.

wilkgr76, 2020-05-21:
Thank you. I'll do that.

closed 2020-05-21 by wilkgr76