armbian/build #1994: compile RetrOrangePi_4.2_FULL_Armbian_5.59

Lyric1984, 2020-05-28:
Hi,
I downloaded a firmware:(http://www.retrorangepi.org/FULL/RetrOrangePi_4.2_FULL_Armbian_5.59_Orangepilite_Debian_jessie_default_3.4.113_desktop.7z) 
I need to modify the kernel, where to download the corresponding source code to compile?

EvilOlaf, 2020-05-28:
Um...ask Retropi for the sources?

Lyric1984, 2020-05-28:
I downloaded the source code of the sunxi-5.3 branch[https://github.com/armbian/build/tree/sunxi-5.3], but the gpu of the system is stuck, which version of the source code should I export, the kernel version is 3.4.113

EvilOlaf, 2020-05-28:
Since this is not an official Armbian image (and this is the Armbian build repository btw. and has nothing to do with RetroPi) I can only guess which sources and maybe patches have been used. Try these but do not expect any support whatsoever. Old legacy has been dropped  by Armbian last year.

https://github.com/armbian/linux

igorpecovnik, 2020-05-28:
... and has nothing to do with this issue tracker, we are closing it.

closed 2020-05-28 by igorpecovnik