armbian/build #2035: KERNEL_CONFIGURE seems to have stopped having any effect

martini1992, 2020-06-18:
Fairly self explanatory, using the KERNEL_CONFIGURE="yes" option in the compile command doesn't cause the cinfig menus to appear during the build. The option is recognised by the script but make is not called with correct switch.

EvilOlaf, 2020-06-18:
Cannot reproduce

igorpecovnik, 2020-06-18:
For me works as expected. Try to remove sources to force download  ... do you use Docker?

martini1992, 2020-06-18:
No i did a fresh checkout with git in a Ubuntu 20.04 VM

igorpecovnik, 2020-06-18:
What if you go directly to the source which was downloaded ... and ran 

`sudo make menuconfig`

For example (Bananapi current) build/cache/sources/linux-mainline/linux-5.6.y

martini1992, 2020-06-18:
Ah ok, firstly I am building for the helios4, and secondly running that command highlighted that menuconfig errors with "Your display is too small to run Menuconfig! It must be at least 19 lines by 80 columns." Which was probably in the build output but was hidden by all the other build output. Would it be possible to cancel the build process if this error occurs, especially since if your asking for the menu then you probably don't want the default kernel config building.

martini1992, 2020-06-18:
After changing the VM window size to allow a larger terminal and rerunning the standard compile command everything works as expected.

I still feel the menuconfig error should stop the build process though.

igorpecovnik, 2020-06-18:
> I still feel the menuconfig error should stop the build process though.

If you have a desire to fix, here is the code:
https://github.com/armbian/build/blob/master/lib/compilation.sh#L352-L384

closed 2020-06-18 by igorpecovnik

martini1992, 2020-06-18:
Cheers, think I've worked out how this can be done. I'll knock something up later.