armbian/build #2048: 2nd partition cannot be created automatically [Pine H64].

jsuchecki, 2020-06-23:
Armbian during first boot resizes root partition but doesn't create a second partition. Built-in service of partitions preparing works longer than 9 minutes with effect as I wrote above.
I noted it in the cases of versions 20.08 and 19.11 (Ubuntu Bionic variant). It is worth attention that I used successful 19.11 in an another project last year. I'm not sure if the issue is related with same Armbian or rather some Ubuntu package.
The image was tested on two different SD cards with the same result. Additional info: partition can be created manually using fdisk.

igorpecovnik, 2020-06-23:
> Armbian during first boot resizes root partition but doesn't create a second partition. 

That is expected behaviour since v1.0. We design it that way. It suited most of the cases, but I am sure not all. In that case you need to dig in and customise the system upon your needs. Check [build documentation](https://docs.armbian.com/Developer-Guide_Build-Preparation/).

ArkadiuszRaj, 2020-06-23:
Hi @igorpecovnik 
So this line:
https://github.com/armbian/build/blob/bbdf0bf269a0a067bfbedd71d12308c139278818/packages/bsp/common/usr/lib/armbian/armbian-resize-filesystem#L117

is some kind of leftovers? And any build system shall not rely on this functionality?

igorpecovnik, 2020-06-23:
Armbian can have separate boot and root but usually not by default. Odroid C1 has 1st partition FAT ... because ext4 driver is unstable and there is no way to change u-boot on this board. So we have a support to extend second partition ... if it exists.

jsuchecki, 2020-06-23:
Thank you for answer. There is still another issue, resizing service starts and works very long (as I said before) on EVERY boot until I'll create second partition manually. It happens despite that root is successful resized during the first time.

jsuchecki, 2020-06-23:
I forgot, please find a boot log there:
https://pastebin.com/pvxyzuxS

It looks like scripts crashes and systemd waits for timeout.

igorpecovnik, 2020-06-23:
>  boot log there

I only see that it is killing the resize ... need to recreate. Which image?

jsuchecki, 2020-06-24:
I built this image using 20.08 scripts and Ubuntu Bionic as a base distribution. I also observed similar behaviour in the case of v. 19.11 so I don't exclude issue in Ubuntu.

igorpecovnik, 2020-06-24:
> I built this image

With the information you gave here, I have absolutely no idea what you did. Build logs or detailed step by step. You build Ubuntu Bionic image with which kernel, desktop, cli, minimal, or you build unknown image on Ubuntu Bionic host. Everything matters!

jsuchecki, 2020-06-25:
Ok, I chose a 'minimal' building mode with some customizing (but rather not important from view of partitions management).
Please find /var/log/armbian-hardware-monitor.log - https://pastebin.com/20gX8TGj
I hope it'll be useful.

igorpecovnik, 2020-06-25:
> with some customizing
> I hope it'll be useful.

It is useful for closing. We can't afford to support your R&D unless you do something for us first:
https://github.com/armbian/build#support

closed 2020-06-25 by igorpecovnik