armbian/build #2077: [ "$EXTRAWIFI" == yes ]  ?

150balbes, 2020-07-03:
&& [ "$EXTRAWIFI" == yes ]

to 

https://github.com/armbian/build/blob/master/lib/compilation-prepare.sh#L100

https://github.com/armbian/build/blob/master/lib/compilation-prepare.sh#L181

EvilOlaf, 2020-07-03:
Well technically the last one is not a WiFi driver ^^.
Maybe rename the flag to `EXTRANETWORK`?

150balbes, 2020-07-03:
IMHO it is not desirable to breed many different flags (the more flags, the harder it is to track them). Just one current one is enough. You need to disable it on new cores that don't have patches yet.

igorpecovnik, 2020-07-03:
> rename the flag

Perhaps adding new one under EXTRANETWORK and disabled by default to minimise troubles.

But both under EXTRAWIFI is also good enough temporally fix.

closed 2020-07-03 by igorpecovnik