armbian/build #2084: HDMI sound in Allwinner A20 (and others)

daniel-sousa-me, 2020-07-07:
The current kernel version has no support for HDMI sound on the Allwinner A20. Stefan from Olimex (Olinuxino) wrote a patch to add support to it, but currently it is only available in LibreElec.

Would you be interested in including the patch in Armbian?

I don't know how to build the image and apply the patch myself, but if someone does it, I can help with the testing.

The patch is available at: https://patchwork.kernel.org/patch/11354381/

EvilOlaf, 2020-07-07:
A20 is sunxi family afair. Put the patch here: https://github.com/armbian/build/tree/master/patch/kernel/sunxi-current

daniel-sousa-me, 2020-07-07:
The hard part for me is building and I don't want to just submit a PR without testing literally anything.

igorpecovnik, 2020-07-07:
> The hard part for me is building and I don't want to just submit a PR without testing literally anything.

Merge CI is testing for building when a merge requests is open. You can try this way .

EvilOlaf, 2020-07-07:
> 
> 
> The hard part for me is building and I don't want to just submit a PR without testing literally anything.

Building is actually pretty straight forward since the script will do all the work for you. Where do you have issues?

tido-, 2020-07-08:
> no support for HDMI sound on the Allwinner A20. Stefan from Olimex (Olinuxino) wrote a patch
This is not from me, but worth to consider:  That driver needs rewrite against slightly different sound framework as requested by Chen-Yu.

igorpecovnik, 2020-07-17:
>  That driver needs rewrite against slightly different sound framework as requested by Chen-Yu.

which can easily take months. Adding support.

closed 2020-07-17 by igorpecovnik

daniel-sousa-me, 2020-07-17:
Thank you very much!

igorpecovnik, 2020-07-17:
You are welcome to test and report. I was not able to do that since I have no access to hw, at least not to hear things. Check beta.armbian.com repository in a few hours - currently rebuilding.

daniel-sousa-me, 2020-07-18:
I will try to test it later today. So I just install the "normal" OS image and then point apt there and upgrade?

igorpecovnik, 2020-07-18:
> So I just install the "normal" OS image and then point apt there and upgrade?

Exactly

... or build manually from the trunk if you have an option.

daniel-sousa-me, 2020-07-19:
With the new kernel, I lose the HDMI output.

I installed the kernel, but now I don't get HDMI output after booting. I see some systemd [done] messages and after a bit the screen just goes blank.

I'll try to look better into it, but I probably won't be able to do that until the middle of the week.

Is there any log that I may send you to help?

igorpecovnik, 2020-07-19:
We just bumped sunxi to 5.4 and 5.7 (current) and release window is closing. No time for any deep analysis at this point. @martinayotte can you check if you can replicate the problem and move the patch out in case it really breaks HDMI video.

daniel-sousa-me, 2020-07-19:
This problem does not appear to be related to the patch. The latest version I could test without this problem was `Linux lime2 5.4.50-sunxi #trunk.30 SMP Wed Jul 1 13:40:39 UTC 2020 armv7l GNU/Linux` and the earliest not working version is `Linux lime2 5.4.51-sunxi #trunk.36 SMP Fri Jul 10 20:55:38 UTC 2020 armv7l GNU/Linux`

igorpecovnik, 2020-07-19:
I think we need to add this https://github.com/OLIMEX/linux-olimex/commit/2689a390c90a181613fcebd6a8965e968cf802d7

daniel-sousa-me, 2020-07-19:
I tried the 5.7 kernels and `Linux lime2 5.7.7-sunxi #trunk.36 SMP Fri Jul 10 21:20:57 UTC 2020 armv7l GNU/Linux` works, but `Linux lime2 5.7.8-sunxi #trunk.38 SMP Tue Jul 14 14:17:49 UTC 2020 armv7l GNU/Linux` and later don't. In this case the signal goes out before the systemd messages start.

It's weird that `Linux lime2 5.7.7-sunxi #trunk.36 SMP Fri Jul 10 21:20:57 UTC 2020 armv7l GNU/Linux` works, but `Linux lime2 5.4.51-sunxi #trunk.36 SMP Fri Jul 10 20:55:38 UTC 2020 armv7l GNU/Linux` doesn't. 

For 5.4, aae4dacbcd654dc7910c43a9425b3b2f8ce00885 appears to be the offending commit, but I don't know what's going on with 5.7

daniel-sousa-me, 2020-07-19:
I had missed a kernel version. `Linux lime2 5.4.51-sunxi #trunk.35 SMP Thu Jul 9 15:46:54 UTC 2020 armv7l GNU/Linux
`doesn't work either`´. It seems that 5.4 broke a full day before 5.7.

igorpecovnik, 2020-07-19:
I can't test HDMI stuff ATM. Out of office ...

There is another interesting commit which could explain it
https://github.com/OLIMEX/linux-olimex/commit/9232f67729b0d4e275de27220e3003ed1a0f5771

martinayotte, 2020-07-19:
@igorpecovnik , on my Olimex-Micro-A20, the HDMI video is working fine with 5.7.7.
I didn't tested sound since the HDMI monitor I've used didn't audio.

daniel-sousa-me, 2020-07-19:
Mine also works with 5.7.7, the problem is with 5.7.8

daniel-sousa-me, 2020-07-21:
I saw your commit and tried again with the new kernel, but I still get the same problem.

On dmesg I see the error `sun4i-drm display-engine: [drm] Cannot find any crtc or sizes`

daniel-sousa-me, 2020-07-21:
I created a virtual machine to compile the kernel (I don't run Ubuntu), but it's still compiling after 3 days

jernejsk, 2020-07-21:
> which can easily take months. Adding support.

It's been half a year already.

daniel-sousa-me, 2020-07-24:
After a lot of tries (and some money) I managed to compile an older version of the kernel and apply the patch, using a machine in Amazon EC2, but I still got no sound :(

I decided to give up and just use Libreelec with a chroot to an Armbian instance.

Thank you very much for your support and for the great distro!

johndoe71rus, 2023-06-04:
I took out the board from the shelf, brushed off the dust.
image Armbian_23.8.0-trunk.51_Cubieboard2_bookworm_edge_6.2.16_minimal.img.xz
no HDMI-audio
[armbian-hardware-monitor.log](https://github.com/armbian/build/files/11646399/armbian-hardware-monitor.log)