armbian/build #2106: N2 mainline ethernet failure, with upstream patch

n2qcn, 2020-07-22:
https://forum.armbian.com/topic/14675-breaking-n2-eth0-in-nightly-and-mainline-patch/

eth0 establishes link on the N2 with released 5.6.18-meson64 (root@builder) but after switching to nightly 5.7.9-meson64 (root@stable) fails to get ethernet link and errors with
 
[   76.174115] meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=25)
[   76.278174] meson8b-dwmac ff3f0000.ethernet: Failed to reset the dma
[   76.278892] meson8b-dwmac ff3f0000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   76.287895] meson8b-dwmac ff3f0000.ethernet eth0: stmmac_open: Hw setup failed
 
I've tracked this to a needed delay in the overlay, a patch is available here:
 
https://forum.khadas.com/t/vim2-ethernet-problem-with-mainline-kernel/5817/7
https://patchwork.kernel.org/patch/10997079/
 
Thanks for an outstanding build!

lanefu, 2020-07-22:
[AR-376]

lanefu, 2020-07-22:
hi @n2qcn thanks for reporting and providing patch.

Test kernel here https://armbian.lane-fu.com/debs-beta/

search for AR-376

lanefu, 2020-07-22:
test image here https://armbian.lane-fu.com/Armbian_20.08.0-trunk-AR-376_Odroidn2_buster_current_5.7.9_minimal.img.xz

n2qcn, 2020-07-22:
Thanks for the build! the error has changed to:
[    2.105712] meson8b-dwmac ff3f0000.ethernet: IRQ eth_wake_irq not found
[    2.105720] meson8b-dwmac ff3f0000.ethernet: IRQ eth_lpi not found
[    2.106113] meson8b-dwmac ff3f0000.ethernet: PTP uses main clock
[    2.106120] meson8b-dwmac ff3f0000.ethernet: no reset control found
[    2.107640] meson8b-dwmac ff3f0000.ethernet: User ID: 0x11, Synopsys ID: 0x37
[    2.107648] meson8b-dwmac ff3f0000.ethernet: 	DWMAC1000
[    2.107654] meson8b-dwmac ff3f0000.ethernet: DMA HW capability register supported
[    2.107658] meson8b-dwmac ff3f0000.ethernet: RX Checksum Offload Engine supported
[    2.107662] meson8b-dwmac ff3f0000.ethernet: COE Type 2
[    2.107665] meson8b-dwmac ff3f0000.ethernet: TX Checksum insertion supported
[    2.107668] meson8b-dwmac ff3f0000.ethernet: Wake-Up On Lan supported
[    2.107694] meson8b-dwmac ff3f0000.ethernet: Normal descriptors
[    2.107698] meson8b-dwmac ff3f0000.ethernet: Ring mode enabled
[    2.107701] meson8b-dwmac ff3f0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[   17.354178] meson8b-dwmac ff3f0000.ethernet eth0: no phy at addr -1
[   17.354186] meson8b-dwmac ff3f0000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -19)
[   17.374470] meson8b-dwmac ff3f0000.ethernet eth0: no phy at addr -1

with the last two repeatable with an `ifconfig eth0 up`:
[  780.133215] meson8b-dwmac ff3f0000.ethernet eth0: no phy at addr -1
[  780.133227] meson8b-dwmac ff3f0000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -19)

root@odroidn2:~# ifconfig -a
eth0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 00:1e:06:42:17:b3  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 6  
[...]
root@odroidn2:~# uname -a
Linux odroidn2 5.7.9-meson64 #376 SMP PREEMPT Tue Jul 21 20:59:27 EDT 2020 aarch64 GNU/Linux

[ note for self: armbian-config is gone but nmtui is listed here to save future google time ]

In my google of this issue I recall a chip reset issue on some boards, I'll try to post that hardware design issue as it might be a contributing factor, even if eth0 works in production builds.

lanefu, 2020-07-22:
oops sorry minimal image dosnt have armbian config, ill build regular next time....

n2qcn, 2020-07-22:
https://community.st.com/s/question/0D50X0000B8iBSB/stm32mp157-ethernet-problem

lanefu, 2020-07-22:
when i downloaded the patch series it gave me the same patch as the first one....

lanefu, 2020-07-22:
oh.. original patch failed to apply :P

lanefu, 2020-07-23:
So Ive got some good news and bad news...

I booted my N2 off that same image, and dmesg has been silent.. even when running iperf3

```
root@odroidn2:~# dmesg|fgrep dwmac
[    2.107658] meson8b-dwmac ff3f0000.ethernet: IRQ eth_wake_irq not found
[    2.107665] meson8b-dwmac ff3f0000.ethernet: IRQ eth_lpi not found
[    2.107756] meson8b-dwmac ff3f0000.ethernet: PTP uses main clock
[    2.107762] meson8b-dwmac ff3f0000.ethernet: no reset control found
[    2.116780] meson8b-dwmac ff3f0000.ethernet: User ID: 0x11, Synopsys ID: 0x37
[    2.116789] meson8b-dwmac ff3f0000.ethernet:         DWMAC1000
[    2.116793] meson8b-dwmac ff3f0000.ethernet: DMA HW capability register supported
[    2.116797] meson8b-dwmac ff3f0000.ethernet: RX Checksum Offload Engine supported
[    2.116800] meson8b-dwmac ff3f0000.ethernet: COE Type 2
[    2.116803] meson8b-dwmac ff3f0000.ethernet: TX Checksum insertion supported
[    2.116805] meson8b-dwmac ff3f0000.ethernet: Wake-Up On Lan supported
[    2.116835] meson8b-dwmac ff3f0000.ethernet: Normal descriptors
[    2.116839] meson8b-dwmac ff3f0000.ethernet: Ring mode enabled
[    2.116842] meson8b-dwmac ff3f0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[   61.973748] meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=25)
[   61.985800] meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
[   61.985811] meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
[   61.985818] meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
[   65.585084] meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
root@odroidn2:~# uname -a
Linux odroidn2 5.7.9-meson64 #376 SMP PREEMPT Tue Jul 21 20:59:27 EDT 2020 aarch64 GNU/Linux
```

n2qcn, 2020-07-23:
well this is embarrassing, Swapped the network switch got link and IP, switched back the original GS108Tv3 switch and got link and IP. sigh, all I needed was a switch power cycle.

lanefu, 2020-07-23:
@n2qcn glad you got it working....  problems like these can always be crazy.  I still appreciate the thoroughness of your research.   Please don't hesitate to report any other issues that come up.

n2qcn, 2020-07-24:
The patch was indeed applied to the AR-376 kernel posted above as this file is present:
/usr/lib/linux-u-boot-current-odroidn2_20.08.0-trunk-AR-376_arm64/u-boot.bin
and if one converts the released to a nightly as stated at the top of this issue, N2's ethernet stills fails to get link with a freshly rebooted network switch, so this patch does indeed fix address the issue.

Pls merge this patch $CURRENT

lanefu, 2020-07-24:
interesting

lanefu, 2020-07-24:
Hmm so at least most of the serious fails.. maybe you can help me figure out what specific patch to keep?

```
Processing file /home/lane/GIT/build/patch/kernel/meson64-current/0101-FROMLIST-Ethernet-PHY-reset-GPIO-updates-for-Amlogic-SoCs.patch
2 out of 2 hunks FAILED -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-g12a-x96-max.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm/boot/dts/meson8b-ec100.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm/boot/dts/meson8b-mxq.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm/boot/dts/meson8b-odroidc1.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm/boot/dts/meson8m2-mxiii-plus.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxbb-nanopi-k2.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxbb-nexbox-a95x.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxbb-odroidc2.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxbb-p200.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxbb-vega-s95.dtsi.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxbb-wetek.dtsi.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxl-s905d-p230.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxm-khadas-vim2.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxm-nexbox-a1.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxm-q200.dts.rej
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-gxm-rbox-pro.dts.rej
1 out of 1 hunk FAILED -- saving rejects to file arch/arm64/boot/dts/amlogic/meson-g12b-odroid-n2.dts.rej
```

lanefu, 2020-07-24:
I think these 2 files...does that line up?

I'll have to look more later

```
> meson-gxbb-vega-s95.dts
> meson-gxbb-wetek.dts
```

n2qcn, 2020-07-24:
I've gone through two network switches and two N2s to make sure I don't have a hardware issue and feel confident the nightly builds on an N2 is a software issue. The great thing is we have a working build you posted above, so I agree we need to isolate what's the different between above and nightly.

meson-gxbb-odroidc2.dts must be my first guess, but based on your patch output we shouldn't rule out other changes to the kernel from nightly.

lanefu, 2020-07-26:
@n2qcn made new build -- wanna try debs-beta and image on https://armbian.lane-fu.com

n2qcn, 2020-07-26:
Both https://armbian.lane-fu.com/Armbian_20.08.0-trunk_Odroidn2_focal_current_5.7.10.img and https://armbian.lane-fu.com/Armbian_20.08.0-trunk-AR-376b_Odroidn2_focal_current_5.7.10.img get link and IP on first boot. 

We almost need an image that fails as the armbian-config process from release to nightly might be interjecting the failure. It  sure loads alot of Debian/Ubuntu packages.

lanefu, 2020-07-26:
release to nights has to change a lot of things

anyway it's safe to say that our next -current kernel when we release 20.08 will work.  still confused as to why you don't get link on other nightlies right now

n2qcn, 2020-07-26:
we agree the next built will work. Pls close this case if we don't want to worry about what is believed to be a release to nightly process issue.

n2qcn, 2020-07-26:
apt update ; apt full-upgrade yields the failure at the top of this issue in https://armbian.lane-fu.com/Armbian_20.08.0-trunk_Odroidn2_focal_current_5.7.10.img which installs:

Unpacking python-apt-common (2.0.0ubuntu0.20.04.1) over (2.0.0) ...
Unpacking python3-apt (2.0.0ubuntu0.20.04.1) over (2.0.0) ...
Unpacking armbian-config (20.08.0-trunk.45) over (20.08.0-trunk) ...
Unpacking armbian-firmware (20.08.0-trunk.45) over (20.08.0-trunk) ...
Unpacking linux-dtb-current-meson64 (20.08.0-trunk.44) over (20.08.0-trunk) ...
Unpacking linux-focal-root-current-odroidn2 (20.08.0-trunk.46) over (20.08.0-trunk) ...
Unpacking linux-image-current-meson64 (20.08.0-trunk.44) over (20.08.0-trunk) ...
Unpacking linux-u-boot-odroidn2-current (20.08.0-trunk.43) over (20.08.0-trunk) ...

while Armbian_20.08.0-trunk-AR-376b_Odroidn2_focal_current_5.7.10.img  only installs

Unpacking python-apt-common (2.0.0ubuntu0.20.04.1) over (2.0.0) ...
Unpacking python3-apt (2.0.0ubuntu0.20.04.1) over (2.0.0) ...

and works after reboot.

lanefu, 2020-07-26:
unclosing

lanefu, 2020-07-26:
okay.... last test... https://armbian.lane-fu.com

cheeseburger-41 images/debs

n2qcn, 2020-07-26:
not sure what the test is here, but like the first Armbian_20.08.0-trunk_Odroidn2_focal_current_5.7.10.img and Armbian_20.08.0-trunk-AR-376b_Odroidn2_focal_current_5.7.10.img , the new Armbian_20.08.0-trunk-cheeseburger41_Odroidn2_focal_current_5.7.10.img all get link and IP. 

cheeseburger41 and AR-376b full-upgrade 

Unpacking python-apt-common (2.0.0ubuntu0.20.04.1) over (2.0.0) ...
Unpacking python3-apt (2.0.0ubuntu0.20.04.1) over (2.0.0) ...

and get link and IP after reboot. But I did see Armbian_20.08.0-trunk_Odroidn2_focal_current_5.7.10.img  was updated (NULL tag and binary file differed)  that I wasn't directed to test does a full-upgrade of:

Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  armbian-config armbian-firmware linux-dtb-current-meson64
  linux-focal-root-current-odroidn2 linux-image-current-meson64
  linux-u-boot-odroidn2-current python-apt-common python3-apt
8 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 49.1 MB of archives.
After this operation, 94.2 kB of additional disk space will be used.
Do you want to continue? [Y/n] Get:1 http://ports.ubuntu.com focal-updates/main arm64 python-apt-common all 2.0.0ubuntu0.20.04.1 [17.4 kB]
Get:2 http://beta.armbian.com focal/main arm64 armbian-config all 20.08.0-trunk.45 [44.0 kB]
Get:3 http://ports.ubuntu.com focal-updates/main arm64 python3-apt arm64 2.0.0ubuntu0.20.04.1 [147 kB]
Get:4 http://beta.armbian.com focal/main arm64 armbian-firmware all 20.08.0-trunk.45 [6,613 kB]
Get:5 http://beta.armbian.com focal/main arm64 linux-dtb-current-meson64 arm64 20.08.0-trunk.44 [205 kB]
Get:6 http://beta.armbian.com focal/main arm64 linux-focal-root-current-odroidn2 arm64 20.08.0-trunk.46 [412 kB]
Get:7 http://beta.armbian.com focal/main arm64 linux-image-current-meson64 arm64 20.08.0-trunk.44 [41.2 MB]
Get:8 http://beta.armbian.com focal/main arm64 linux-u-boot-odroidn2-current arm64 20.08.0-trunk.43 [493 kB]
Fetched 49.1 MB in 4s (12.0 MB/s)
(Reading database ... 35319 files and directories currently installed.)
Preparing to unpack .../0-python-apt-common_2.0.0ubuntu0.20.04.1_all.deb ...
Unpacking python-apt-common (2.0.0ubuntu0.20.04.1) over (2.0.0) ...
Preparing to unpack .../1-python3-apt_2.0.0ubuntu0.20.04.1_arm64.deb ...
Unpacking python3-apt (2.0.0ubuntu0.20.04.1) over (2.0.0) ...
Preparing to unpack .../2-armbian-config_20.08.0-trunk.45_all.deb ...
Unpacking armbian-config (20.08.0-trunk.45) over (20.08.0-trunk) ...
Preparing to unpack .../3-armbian-firmware_20.08.0-trunk.45_all.deb ...
Unpacking armbian-firmware (20.08.0-trunk.45) over (20.08.0-trunk) ...
Preparing to unpack .../4-linux-dtb-current-meson64_20.08.0-trunk.44_arm64.deb ...
Unpacking linux-dtb-current-meson64 (20.08.0-trunk.44) over (20.08.0-trunk) ...
Preparing to unpack .../5-linux-focal-root-current-odroidn2_20.08.0-trunk.46_arm64.deb ...
Unpacking linux-focal-root-current-odroidn2 (20.08.0-trunk.46) over (20.08.0-trunk) ...
Preparing to unpack .../6-linux-image-current-meson64_20.08.0-trunk.44_arm64.deb ...
Unpacking linux-image-current-meson64 (20.08.0-trunk.44) over (20.08.0-trunk) ...
Preparing to unpack .../7-linux-u-boot-odroidn2-current_20.08.0-trunk.43_arm64.deb ...
Unpacking linux-u-boot-odroidn2-current (20.08.0-trunk.43) over (20.08.0-trunk) ...
Setting up armbian-config (20.08.0-trunk.45) ...
Setting up linux-image-current-meson64 (20.08.0-trunk.44) ...
update-initramfs: Generating /boot/initrd.img-5.7.10-meson64
update-initramfs: Converting to u-boot format
Setting up linux-dtb-current-meson64 (20.08.0-trunk.44) ...
Setting up linux-u-boot-odroidn2-current (20.08.0-trunk.43) ...
Setting up linux-focal-root-current-odroidn2 (20.08.0-trunk.46) ...
NOTE: You can find previous bootscript versions in /usr/share/armbian !
Recreating boot script
Setting up armbian-firmware (20.08.0-trunk.45) ...
Setting up python-apt-common (2.0.0ubuntu0.20.04.1) ...
Setting up python3-apt (2.0.0ubuntu0.20.04.1) ...
Processing triggers for initramfs-tools (0.136ubuntu6.2) ...
update-initramfs: Generating /boot/initrd.img-5.7.10-meson64
update-initramfs: Converting to u-boot format

and yields the same error at top of post with no ethernet link and no IP.

Perhaps what ever is in http://beta.armbian.com is not trunk and still contains whatever unidentified issues that causes the loss of ethernet link and once merged with your trunk all will be good in the world?

n2qcn, 2020-07-27:
for http://armbian.lane-fu.com/Armbian_20.08.0-trunk_Odroidn2_focal_current_5.7.10.img who's ethernet gets link and IP the apt upgrade package that causes the ethernet to fail is: 

http://beta.armbian.com/pool/main/l/linux-focal-root-current-odroidn2/linux-focal-root-current-odroidn2_20.08.0-trunk.48_arm64.deb

update it with what's in your current build and we have world peace. This results in a broken ethernet:

root@odroidn2:/home/rob# uname -a
Linux odroidn2 5.7.10-meson64 #trunk.47 SMP PREEMPT Mon Jul 27 06:08:55 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
root@odroidn2:/home/rob# apt list --upgradable -a
Listing... Done
linux-focal-root-current-odroidn2/focal 20.08.0-trunk.48 arm64 [upgradable from: 20.08.0-trunk]
linux-focal-root-current-odroidn2/focal 20.08.0-trunk.47 arm64
linux-focal-root-current-odroidn2/focal 20.08.0-trunk.46 arm64
linux-focal-root-current-odroidn2/focal 20.08.0-trunk.45 arm64
linux-focal-root-current-odroidn2/focal 20.08.0-trunk.44 arm64
linux-focal-root-current-odroidn2/focal 20.08.0-trunk.43 arm64
linux-focal-root-current-odroidn2/now 20.08.0-trunk arm64 [installed,upgradable to: 20.08.0-trunk.48]

root@odroidn2:/home/rob# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  linux-focal-root-current-odroidn2
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 412 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://beta.armbian.com focal/main arm64 linux-focal-root-current-odroidn2 arm64 20.08.0-trunk.48 [412 kB]
Fetched 412 kB in 1s (455 kB/s)                            
(Reading database ... 35320 files and directories currently installed.)
Preparing to unpack .../linux-focal-root-current-odroidn2_20.08.0-trunk.48_arm64.deb ...
Unpacking linux-focal-root-current-odroidn2 (20.08.0-trunk.48) over (20.08.0-trunk) ...
Setting up linux-focal-root-current-odroidn2 (20.08.0-trunk.48) ...
NOTE: You can find previous bootscript versions in /usr/share/armbian !
Recreating boot script
Processing triggers for initramfs-tools (0.136ubuntu6.2) ...
update-initramfs: Generating /boot/initrd.img-5.7.10-meson64
update-initramfs: Converting to u-boot format

n2qcn, 2020-07-27:
This patch does about the same as the top patch, but is isolated to meson-gxm-khadas-vim2.dts 
If we prefer to merge just what's needed for this issues.

https://github.com/khadas/fenix/blob/master/packages/linux-mainline/patches/5.7/0086-fix-broken-ethernet-interface-up-down.patch

note: ifconfig eth0 down ; ifconfig up yields the same error at the top of the issue with the cheeseburger41 and other images and is likely a better unit test for this issue than a reboot.

lanefu, 2020-07-27:
so that cheeseburger-41 image and debs contained no additional patches.. just all the stuff slated for next release....

n2qcn, 2020-07-29:
ifconfig eth0 down ; ifconfig eth0 up on released 5.6.18 yields the same error at the top of this issue.

It would be interesting to see if an N2 overlay without reset-gpios = <&gpio GPIOZ_14 GPIO_ACTIVE_LOW>;
would fix this issue. (Sigh, I really should learn how to build overlays)

n2qcn, 2020-07-31:
The N2+ with cheeseburger yields the errors at the top of this issue on bootup.

Here is the fix:

dtc -I dtb -O dts /boot/dtb/amlogic/meson-g12b-odroid-n2.dtb | grep -v 'reset-gpios = <0x21 0x0f 0x07>;' > /tmp/t
dtc -I dts -O dtb /tmp/t > /boot/dtb/amlogic/meson-g12b-odroid-n2.dtb

This works with an N2 and N2+ in 5.6.18 and 5.7.10 with the acid test of `ifconfig eth0 down ; ifconfig eth0 up`

I tried to write this as an overlay but failed. With alittle guidance (pointers to the git repo with the source of /boot/dtb/amlogic/meson-g12b-odroid-n2.dtb) I will research the proper way to make an overlay or a an armbiad build patch.

igorpecovnik, 2020-08-28:
This seems to work: https://github.com/armbian/build/pull/2166 Tested 3x - upgrading to latest build, works ok.

DarthGandalf, 2020-11-14:
The patch from #2166 breaks the network for me on N2+.
```
[   17.568439] meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=34)
[   17.675393] meson8b-dwmac ff3f0000.ethernet: Failed to reset the dma
[   17.675399] meson8b-dwmac ff3f0000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   17.675401] meson8b-dwmac ff3f0000.ethernet eth0: stmmac_open: Hw setup failed
```

Removing the patch (putting `reset-gpios = <&gpio GPIOZ_15 (GPIO_ACTIVE_LOW | GPIO_OPEN_DRAIN)>;` back) fixed it.

Using kernel 5.9.8 with patches and config as of a08539790d767029ce404f543f5537d62b99ec7f

Jerome-Maurin, 2020-11-17:
With the `reset-gpios = <&gpio GPIOZ_15 (GPIO_ACTIVE_LOW | GPIO_OPEN_DRAIN)>;` in the device-tree on my N2, on some occasions I end up with the same issue as @n2qcn.

No interface at startup or `ip down` followed by `ip up` fails.

The only way I found in my case to end up with something stable, was removing the `reset-gpios` line AND to add a ugly fix in `/etc/network/interfaces` in my case (using `ifupdown` for the network configuration):
```
auto eth0
iface eth0 inet dhcp
# Ugly fix to make ethernet work (able to up & down as pleased), works with DT patch
  pre-up ip link set dev eth0 up || true
```
When the system starts with a device-tree without the `reset-gpios`, you sometime en up with your issue @DarthGandalf.
The fix is to up the interface a second time (the theory is that the first time resets the hardware on failure which put the hardware back in the correct state for the second time to work).

My ugly fix just adds and `ip link set up` (would also work with `ifconfig`) before the normal up done by `ifupdown`.

I consider all this as temporary fix.

There definitely is a real issue somewhere, whether it is a hardware one on some revision of the Odroid N2, or a software one in the driver code.

I don't have neither the time nor the expertise at the moment to truly take some time to address this, so it will have to do for now as I'm only using this board for personal use.

Maybe some kernel developer could take a serious look at this.

This fix has been stable on my N2 for the last few months with 5.6, 5.7 and 5.8 kernels.

agners, 2020-12-01:
FYI, as I run into this thread when googling: We run into the same issue in Home Assistant OS, see https://github.com/home-assistant/operating-system/issues/1038. The problem is the PHY requires at least 50ms after asserting reset before talking to it. Sent the patch also upstream: https://lkml.org/lkml/2020/11/30/1443

lanefu, 2020-12-01:
Going to re-open issue

lanefu, 2020-12-01:
trying to figure out if this. is a situation where a patch that fixes 1 device breaks another.

lanefu, 2020-12-01:
@Tonymac32 FYI re-opening and will explore this patch

lanefu, 2020-12-01:
[AR-554]

lanefu, 2020-12-01:
Test Image here: https://armbian.lane-fu.com/Armbian_21.02.0-trunk-AR-554_Odroidn2_focal_current_5.9.11.img.xz

n2qcn, 2020-12-01:
I've been using armbian's dtb for my N2 and N2+ with positive results.

As a control I tested current 5.9.6 Focal and dist-upgrade to 5.9.10 on a N2+ and both found an IP and survived ifconfig up ; down.

The image above does not boot for me using U-Boot 2015.01 (Nov 30 2020 - 20:44:39) with:
NOT EXIST, now build display.bin
File System is consistent
Invalid path
** Error ext4fs_write() **
** Unable to write file /boot/display.bin **
** File not found /boot/armbianEnv.txt **
** Unrecognized filesystem type **
113 bytes read in 4 ms (27.3 KiB/s)
2469440 bytes read in 700 ms (33.9 MiB/s)
7527 bytes read in 15 ms (5 MiB/s)
1494220 bytes read in 1506 ms (9.5 MiB/s)
ee_gate_off ...
## Booting kernel from Legacy Image at 01100000 ...
   Image Name:   Linux
   Image Type:   AArch64 Linux Kernel Image (uncompressed)
   Data Size:    2469376 Bytes = 23.7 MiB
   Load Address: 0100000
   Entry Point:  0100000
   Verifying Checksum ... OK
Wrong Ramdisk Image Format
Ramdisk image is corrupt or invalid
[rsvmem] get fdtaddr NULL!

lanefu, 2020-12-01:
blerg thanks for testing...  we've got some unrelated woes with Amlogic right now...

lanefu, 2020-12-01:
For further discussion

https://patchwork.kernel.org/project/linux-amlogic/patch/14754fd95378b78eb9a0a3f8b6bab13f7263c7f1.1606828668.git.stefan@agner.ch/

Jerome-Maurin, 2020-12-02:
> 
> 
> FYI, as I run into this thread when googling: We run into the same issue in Home Assistant OS, see [home-assistant/operating-system#1038](https://github.com/home-assistant/operating-system/issues/1038). The problem is the PHY requires at least 50ms after asserting reset before talking to it. Sent the patch also upstream: https://lkml.org/lkml/2020/11/30/1443

GG.

Hi, tested your patch on a 5.8 Debian Kernel, it seems to be working fine !

Waiting eagerly for your patch to be mainlined.

Thanks for the hard work.

igorpecovnik, 2021-01-27:
Bug not present anymore.

closed 2021-01-27 by igorpecovnik