armbian/build #2111: Problems with compile.sh

Threefish4096, 2020-07-23:
I have tried several times and with different versions to compile my image for the bananapi r2 device, everything has always gone well, now after the 20.05 release it seems that it is no longer able to compile my image.

[image]

With version 20.02 I have the problem that 'buster' doesn't work, and that's fine, it fits.

[image]

However, the problem also exists with version 20.05.
It does not give the same error but crashes during compilation to the following file.

As for the nightly or the version that I download directly from the git via the 'git clone https://github.com/armbian/build.git' command, I get the following problem.

The image is compiled, but once inserted on the micro SD as I have always done, the device does not start and shows no signs of life.
[image]

The command that I have always used in all compilations, even when everything worked, is the following: "sudo ./compile.sh BOARD ="bananapir2" BRANCH ="legacy" RELEASE ="buster" BOARD_NAME ="Banana Pi R2" BOARDFAMILY ="mt7623" BOOTCONFIG ="mt7623n_bpir2_defconfig" BUILD_DESKTOP ="no" KERNEL_TARGET ="legacy" KERNEL_ONLY ="no" KERNEL_CONFIGURE ="no" KERNEL_KEEP_CONFIG ="no" BUILD_MINIMAL="yes" EXTERNAL="no" BSPFREEZE="no" INSTALL_HEADERS="no" EXTERNAL_NEW="no" CREATE_PATCHES="no" CLEAN_LEVEL="make,debs,images,alldebs,cache,oldcache,sources,extras"

Could you help me? Many thanks in advance!

Roman3349, 2020-07-23:
I have the same problem.

**Snippet with error messages**:
```
cc1: error: unrecognized -mtune target: native
cc1: note: valid arguments are: arm2 arm250 arm3 arm6 arm60 arm600 arm610 arm620 arm7 arm7d arm7di arm70 arm700 arm700i arm710 arm720 arm710c arm7100 arm7500 arm7500fe arm7m arm7dm arm7dmi arm8 arm810 strongarm strongarm110 strongarm1100 strongarm1110 fa526 fa626 arm7tdmi arm7tdmi-s arm710t arm720t arm740t arm9 arm9tdmi arm920 arm920t arm922t arm940t ep9312 arm10tdmi arm1020t arm9e arm946e-s arm966e-s arm968e-s arm10e arm1020e arm1022e xscale iwmmxt iwmmxt2 fa606te fa626te fmp626 fa726te arm926ej-s arm1026ej-s arm1136j-s arm1136jf-s arm1176jz-s arm1176jzf-s mpcorenovfp mpcore arm1156t2-s arm1156t2f-s cortex-m1 cortex-m0 cortex-m0plus cortex-m1.small-multiply cortex-m0.small-multiply cortex-m0plus.small-multiply generic-armv7-a cortex-a5 cortex-a7 cortex-a8 cortex-a9 cortex-a12 cortex-a15 cortex-a17 cortex-r4 cortex-r4f cortex-r5 cortex-r7 cortex-r8 cortex-m7 cortex-m4 cortex-m3 marvell-pj4 cortex-a15.cortex-a7 cortex-a17.cortex-a7 cortex-a32 cortex-a35 cortex-a53 cortex-a57 cortex-a72 cortex-a73 exynos-m1 xgene1 cortex-a57.cortex-a53 cortex-a72.cortex-a53 cortex-a73.cortex-a35 cortex-a73.cortex-a53 cortex-a55 cortex-a75 cortex-a75.cortex-a55 cortex-m23 cortex-m33 cortex-r52
cc1: error: unrecognized -mtune target: native
cc1: note: valid arguments are: arm2 arm250 arm3 arm6 arm60 arm600 arm610 arm620 arm7 arm7d arm7di arm70 arm700 arm700i arm710 arm720 arm710c arm7100 arm7500 arm7500fe arm7m arm7dm arm7dmi arm8 arm810 strongarm strongarm110 strongarm1100 strongarm1110 fa526 fa626 arm7tdmi arm7tdmi-s arm710t arm720t arm740t arm9 arm9tdmi arm920 arm920t arm922t arm940t ep9312 arm10tdmi arm1020t arm9e arm946e-s arm966e-s arm968e-s arm10e arm1020e arm1022e xscale iwmmxt iwmmxt2 fa606te fa626te fmp626 fa726te arm926ej-s arm1026ej-s arm1136j-s arm1136jf-s arm1176jz-s arm1176jzf-s mpcorenovfp mpcore arm1156t2-s arm1156t2f-s cortex-m1 cortex-m0 cortex-m0plus cortex-m1.small-multiply cortex-m0.small-multiply cortex-m0plus.small-multiply generic-armv7-a cortex-a5 cortex-a7 cortex-a8 cortex-a9 cortex-a12 cortex-a15 cortex-a17 cortex-r4 cortex-r4f cortex-r5 cortex-r7 cortex-r8 cortex-m7 cortex-m4 cortex-m3 marvell-pj4 cortex-a15.cortex-a7 cortex-a17.cortex-a7 cortex-a32 cortex-a35 cortex-a53 cortex-a57 cortex-a72 cortex-a73 exynos-m1 xgene1 cortex-a57.cortex-a53 cortex-a72.cortex-a53 cortex-a73.cortex-a35 cortex-a73.cortex-a53 cortex-a55 cortex-a75 cortex-a75.cortex-a55 cortex-m23 cortex-m33 cortex-r52
make[4]: *** [scripts/Makefile.build:266: drivers/net/wireless/rtl8812au/core/rtw_security.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [scripts/Makefile.build:266: drivers/net/wireless/rtl8812au/core/rtw_cmd.o] Error 1
make[3]: *** [scripts/Makefile.build:500: drivers/net/wireless/rtl8812au] Error 2
make[3]: *** Waiting for unfinished jobs....
```

EvilOlaf, 2020-07-23:
Please use `./compile.sh` and read the warning when selecting CSC/EOS boards again.

Roman3349, 2020-07-23:
It is caused by `-mtune=native` in Makefile of RTL8812AU/21AU and RTL8814AU driver. See https://github.com/aircrack-ng/rtl8812au/commit/52f38dd3840117d747dcce4149cd30a451774cf8#diff-b67911656ef5d18c4ae36cb6741b7965.

EvilOlaf, 2020-07-23:
The RTL8812AU is currently fixed to a certain commit since newer once broke compilation for unknown reason. Maybe you nailed it.
https://github.com/armbian/build/blob/74dcb00569f515e9b63965c721241a03968671cc/lib/compilation-prepare.sh#L264

Is suggested to revist this by time: https://armbian.atlassian.net/projects/AR/issues/AR-380

Threefish4096, 2020-07-25:
> 
> 
> Please use `./compile.sh` and read the warning when selecting CSC/EOS boards again.
> 
> Btw. compiling flawless for me.
> `./compile.sh BOARD=bananapir2 BRANCH=legacy BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=no`

I try your command and all run with the "nightly", thank you for now!

Threefish4096, 2020-07-27:
Sorry for the inconvenience, but after filling in the image correctly, I can't log in via SSH, I put username: root and password: 1234, but it won't let me in, can someone help me?

P.S is as if expecting a public key

[image]

[image]

[image]

igorpecovnik, 2020-07-27:
>  read the warning when selecting CSC/EOS boards again.

Have you read what does that mean? This means we don't deal with this board and we don't provide any support for it. Its a community contribution - seek help here https://forum.armbian.com/forum/25-peer-to-peer-technical-support/

> can someone help me?

With what? Who?
https://github.com/armbian/build#support

Please use forum for support. Initial issue was resolved, closing.

closed 2020-07-27 by igorpecovnik