armbian/build #2120: smp_affinity: ambiguous redirect

dagix5, 2020-07-30:
In the boot log of Armbian on NanoPi M4 2GB, the are the following errors:
```
Jul 30 16:34:15 localhost armbian-hardware-optimization[795]: /usr/lib/armbian/armbian-hardware-optimization: line 183: /proc/irq/$(awk -F":" "/ehci/ {print \$1}" </proc/interrupts | sed 's/\ //g')/smp_affinity:
 ambiguous redirect
Jul 30 16:34:15 localhost armbian-hardware-optimization[795]: /usr/lib/armbian/armbian-hardware-optimization: line 184: /proc/irq/$(awk -F":" "/ohci/ {print \$1}" </proc/interrupts | sed 's/\ //g')/smp_affinity:
 ambiguous redirect
Jul 30 16:34:15 localhost armbian-hardware-optimization[795]: /usr/lib/armbian/armbian-hardware-optimization: line 185: /proc/irq/$(awk -F":" "/xhci/ {print \$1}" </proc/interrupts | sed 's/\ //g')/smp_affinity:
 ambiguous redirect
Jul 30 16:34:15 localhost armbian-hardware-optimization[795]: /usr/lib/armbian/armbian-hardware-optimization: line 186: /proc/irq/$(awk -F":" "/eth0/ {print \$1}" </proc/interrupts | sed 's/\ //g')/smp_affinity:
 No such file or directory
```
This is my `/proc/interrupts`:
```
           CPU0       CPU1       CPU2       CPU3       CPU4       CPU5       
 10:          0          0          0          0          0          0     GICv3  25 Level     vgic
 12:          0          0          0          0          0          0     GICv3  27 Level     kvm guest vtimer
 15:      40124      42883      40065      36698     155185     139308     GICv3  30 Level     arch_timer
 17:          0          0          0          0          0          0     GICv3 113 Level     rk_timer
 18:          0          0          0          0          0          0  GICv3-23   0 Level     arm-pmu
 19:          0          0          0          0          0          0  GICv3-23   1 Level     arm-pmu
 20:          0          0          0          0          0          0     GICv3  37 Level     ff6d0000.dma-controller
 21:          0          0          0          0          0          0     GICv3  38 Level     ff6d0000.dma-controller
 22:          0          0          0          0          0          0     GICv3  39 Level     ff6e0000.dma-controller
 23:          0          0          0          0          0          0     GICv3  40 Level     ff6e0000.dma-controller
 24:          2          0          0          0          0          0     GICv3  81 Level     pcie-sys
 26:         28      24932          0          0          0          0     GICv3  83 Level     pcie-client
 28:     202697       4532          0          0          0          0     GICv3  96 Level     dw-mci
 29:      16793      47190          0          0          0          0     GICv3  97 Level     dw-mci
 30:         52          0          0          0          0          0     GICv3  43 Level     mmc2
 31:          0          0          0          0          0          0     GICv3  58 Level     ehci_hcd:usb1
 32:          0          0          0          0          0          0     GICv3  60 Level     ohci_hcd:usb3
 33:          0          0          0          0          0          0     GICv3  62 Level     ehci_hcd:usb2
 34:          0          0          0          0          0          0     GICv3  64 Level     ohci_hcd:usb4
 35:          1          0          0          0          0          0     GICv3  94 Level     ff100000.saradc
 36:         36          0          0          0          0          0     GICv3  91 Level     ff110000.i2c
 37:          0          0          0          0          0          0     GICv3  67 Level     ff120000.i2c
 38:          0          0          0          0          0          0     GICv3  68 Level     ff160000.i2c
 39:        104          0          0          0          0          0     GICv3 131 Level     ttyS0
 40:          5          0          0          0          0          0     GICv3 132 Level     ttyS2
 41:          0          0          0          0          0          0     GICv3 129 Level     rockchip_thermal
 42:      22908          0          0          0          0          0     GICv3  89 Level     ff3c0000.i2c
 43:        213          0          0          0          0          0     GICv3  88 Level     ff3d0000.i2c
 44:          0          0          0          0          0          0     GICv3 146 Level     ff650000.video-codec
 45:          0          0          0          0          0          0     GICv3 145 Level     ff650000.video-codec
 46:          0          0          0          0          0          0     GICv3 147 Level     ff650800.iommu
 47:          0          0          0          0          0          0     GICv3  87 Level     ff680000.rga
 52:          0          0          0          0          0          0     GICv3 151 Level     ff8f3f00.iommu, ff8f0000.vop
 53:          0          0          0          0          0          0     GICv3 150 Level     ff903f00.iommu, ff900000.vop
 54:          0          0          0          0          0          0     GICv3  75 Level     ff914000.iommu
 55:          0          0          0          0          0          0     GICv3  76 Level     ff924000.iommu
 56:          0          0          0          0          0          0     GICv3  55 Level     ff940000.hdmi, dw-hdmi-cec
 57:          0          0          0          0          0          0     GICv3  52 Level     job
 58:          0          0          0          0          0          0     GICv3  53 Level     mmu
 59:          2          0          0          0          0          0     GICv3  51 Level     gpu
 70:          1          0          0          0          0          0  rockchip_gpio_irq   5 Edge      GPIO Key Power
 72:          0          0          0          0          0          0  rockchip_gpio_irq   7 Edge      fe320000.dwmmc cd
 99:          2          0          0          0          0          0  rockchip_gpio_irq   2 Level     fsc_interrupt_int_n
118:          0          0          0          0          0          0  rockchip_gpio_irq  21 Level     rk808
225:          0          0          0          0          0          0     GICv3  59 Level     rockchip_usb2phy
226:          0          0          0          0          0          0     GICv3 135 Level     rockchip_usb2phy_bvalid
227:          0          0          0          0          0          0     GICv3  63 Level     rockchip_usb2phy
229:          0          0          0          0          0          0   ITS-MSI   0 Edge      PCIe PME, aerdrv
230:      14876          0          0          0          0          0   ITS-MSI 524288 Edge      ahci[0000:01:00.0]
231:          0          0          0          0          0          0     GICv3 137 Level     dwc3-otg, xhci-hcd:usb5
232:    1578540          0          0          0          0          0     GICv3 142 Level     xhci-hcd:usb7
238:          0          0          0          0          0          0     rk808   5 Edge      RTC alarm
IPI0:     39601      62220      53128     258342     550883     543434       Rescheduling interrupts
IPI1:       689        657        671        616       9914       9049       Function call interrupts
IPI2:         0          0          0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0          0          0       CPU stop (for crash dump) interrupts
IPI4:         0          0          0          0          0          0       Timer broadcast interrupts
IPI5:       982        511        418        578       1623       1500       IRQ work interrupts
IPI6:         0          0          0          0          0          0       CPU wake-up interrupts
Err:          0

```

Versions:
```
armbian-config/buster,buster,now 20.05.9 all [installed]
armbian-firmware/buster,buster,now 20.05.7 all [installed]
```

Please let me know if you need other info.

Thanks!

EvilOlaf, 2020-07-30:
This is the bug tracker for the build script, not for the OS. For last please refer to the forums: https://forum.armbian.com/

dagix5, 2020-07-30:
> This is the bug tracker for the build script, not for the OS. For last please refer to the forums: https://forum.armbian.com/

The bug (if there is one) is in the script `armbian-hardware-optimization` in this repo.

igorpecovnik, 2020-07-30:
Check support terms https://github.com/armbian/build#support and please read them twice.

> The bug (if there is one) is in the script armbian-hardware-optimization in this repo.

Build tools obviously build things correctly but yes its possible that the bug is in some scripts or a patch that is a part of this repository. Since there is 1000x more bugs than resources we don't even try to collect bugs. Its is impossible to explain people that five people can't fix everything ...

Even the bug is real, its not critical and there is nobody who will take it and resolve anytime soon. Go to forum and provide logs we request - symptoms are far too little to get started.

closed 2020-07-30 by igorpecovnik