armbian/build #2136: Docker build fails: /bin/bash misisng in chroot

Legogris, 2020-08-06:
Trying to Docker build bullseye for odroidc4 fails.

Version: Recent `master` (`b4e23b5b6a4ac679d07765ddc057562adb9f223b`)

```
$ ./compile.sh docker BOARD=odroidc4 BRANCH=current CLEAN_LEVEL='make,debs,cache,sources' BUILD_DESKTOP=no RELEASE=bullseye

[...]
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Up to date
[ o.k. ] Checking git sources [ linux-firmware-git master ]
[ .... ] Up to date
[ o.k. ] Creating board support package [ odroidc4 current ]
[ .... ] Fingerprinting
[ o.k. ] Building package [ linux-bullseye-root-current-odroidc4 ]
[ o.k. ] Building desktop package [ armbian-bullseye-desktop_20.08.0-trunk_all ]
[ o.k. ] Starting rootfs and image building process for [ odroidc4 bullseye ]
[ o.k. ] Checking for local cache [ bullseye-cli-arm64.ab3...cad.tar.lz4 ]
[ o.k. ] Extracting bullseye-cli-arm64.ab3...cad.tar.lz4 [ 0 days old ]
[ .... ] bullseye-cli-arm64.ab3...cad.tar.lz4:  469MiB [ 240MiB/s] [==========================================================================================================================================>] 100%
[ o.k. ] Adding Armbian repository and authentication key [ /etc/apt/sources.list.d/armbian.list ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ o.k. ] Applying distribution specific tweaks for [ bullseye ]
[ o.k. ] Applying common tweaks
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ .... ] Installing [ linux-u-boot-current-odroidc4_20.08.0-trunk_arm64.deb ]
[ .... ] Installing [ linux-image-current-meson64_20.08.0-trunk_arm64.deb ]
[ .... ] Installing [ linux-dtb-current-meson64_20.08.0-trunk_arm64.deb ]
[ .... ] Installing [ armbian-firmware_20.08.0-trunk_all.deb ]
[ .... ] Installing [ armbian-config_20.08.0-trunk_all.deb ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ o.k. ] Enabling serial console [ ttyAML0 ]
cp: cannot stat '/root/armbian/.tmp/rootfs-current-odroidc4-bullseye-no-no/etc/armbian-release': No such file or directory
[ o.k. ] Building kernel splash logo [ bullseye ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ .... ] Installing additional packages [ EXTERNAL_NEW ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ o.k. ] Calling image customization script [ customize-image.sh ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ error ] ERROR in function customize_image [ image-helpers.sh:119 ]
[ error ] customize-image.sh exited with error (rc: 127)
[ o.k. ] Process terminated
[ o.k. ] Unmounting [ /root/armbian/.tmp/rootfs-current-odroidc4-bullseye-no-no/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:66 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

No prior warnings or errors in stderr or stdout.

igorpecovnik, 2020-08-06:
You have something wrong with your Docker installation. I just checked on my workstation, where I didn't have any Docker install, and things works. Since I use latest Linux mint, auto installer wasn't successful. Remove container and do it again:

    ./compile.sh dockerpurge

Legogris, 2020-08-06:
@igorpecovnik Doing this `bullseye` doesn't show up as an option and `dockerpurge` seems to ignore the `RELEASE` arg - maybe I'm missing something?

igorpecovnik, 2020-08-06:
> Doing this bullseye doesn't show up as an option 

Because Bullseye is not yet supported and is only available for developers. Add EXPERT="yes" parameter.

dockerpurge is here only to kill docker container which you can also do manually. You have to delete Armbian Docker image somehow since the one you have looks broken.

> maybe I'm missing something?

https://docs.armbian.com/Developer-Guide_Build-Options/#hidden-options-for-advanced-users-default-values-are-marked-bold

Legogris, 2020-08-06:
@igorpecovnik 

Got it. Same result, made sure no armbian-related images or containers existed before trying again.

igorpecovnik, 2020-08-06:
What is your host system? And can you start from a clean git clone that no caches exits.

Legogris, 2020-08-06:
@igorpecovnik I just did, and same results there as well.

Arch Linux.

Linux kernel `5.7.11-arch1-1` on x86_64.

```
$ docker version
Client:
 Version:           19.03.12-ce
 API version:       1.40
 Go version:        go1.14.5
 Git commit:        48a66213fe
 Built:             Sat Jul 18 01:33:21 2020
 OS/Arch:           linux/amd64
 Experimental:      true

Server:
 Engine:
  Version:          19.03.12-ce
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.14.5
  Git commit:       48a66213fe
  Built:            Sat Jul 18 01:32:59 2020
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          v1.3.4.m
  GitCommit:        d76c121f76a5fc8a462dc64594aea72fe18e1178.m
 runc:
  Version:          1.0.0-rc91
  GitCommit:        24a3cf88a7ae5f4995f6750654c0e2ca61ef4bb2
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
```

Stupid question, but without looking through the script very closely, shouldn't all this be done in docker and therefore host environment shouldn't have an impact apart from cpu arch? What other things can vary that I should consider checking?

igorpecovnik, 2020-08-06:
> What other things can vary that I should consider checking?

Did you try to build something else, like Bionic or Focal?

Legogris, 2020-08-06:
@igorpecovnik Get the same for `buster` and `bionic`. Could it be qemu-raleted perhaps?

igorpecovnik, 2020-08-06:
Running low on ideas ... In case Docker under random Linux doesn't work -> full virtualisation would be by far fastest method.

Legogris, 2020-08-07:
Same commit and config builds fine in vagrant, never figure out what was going on in Docker. Cheers @igorpecovnik

sjthearts, 2020-10-29:
@igorpecovnik I am experiencing this issue on Ubuntu 20 VM on MacOS Parallels.  I am debugging, but do you have any specific diagnostic requests?

EvilOlaf, 2020-10-29:
Just curios. Why do you need/use `NO_HOST_CHECK=yes` when using a virtual machine with Focal? Does the compilation abort without it?

sjthearts, 2020-10-29:
I pasted from the wrong window, problem in chair.  Edited.

EvilOlaf, 2020-10-29:
Ah well...
Anyways. Using Parallels is mostly pretty close to using virtulabox to emulate a native environment which makes this odd...
Two things I remember. Ubuntu somtimes installing a "cloud" linux image which is missing binmft kernel module. Workaround is making sure to use a "generic" kernel image.
The other thing I like to mention is the `output/debug/` folder which contains all information collected during building including errors. Start digging there.

EvilOlaf, 2021-03-26:
Assuming solved.

closed 2021-03-26 by EvilOlaf