armbian/build #2141: Encrypted build is not working

Elbandi, 2020-08-10:
I build armbian encrypted fs from 869a89d6e996cda4a628c548f3981c03d1f77698 with following command:

    ./compile.sh  BOARD=bananapim2ultra BRANCH=current RELEASE=bionic BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE=kukac

after it written to sdcard, boot but cannot decrypt the rootfs. I got lvm is not available. after i got revovery prompt, i checked /dev, and no sda nether mmcblk0 devices (internal emmc storage). So look like somethink is missing from kernel.

igorpecovnik, 2020-08-10:
> I build armbian encrypted fs 

Did you check documentation?
https://docs.armbian.com/Developer-Guide_Build-Options/

Warning: This feature was added as community contribution and mostly functional. Under some circumstances though the prompt will not be shown. Therefore it should be considered experimental. Check [here](https://github.com/armbian/build/commit/681e58b6689acda6a957e325f12e7b748faa8330) and [here](https://github.com/armbian/build/issues/1183)

> bananapim2ultra

https://www.armbian.com/download/?tx_maker=sinovoip
CSC

NO OFFICIAL SUPPORT (CSC) are devices which are supported by the 3rd party. We usually don’t have hardware samples nor conduct any tests but they might work. There is no support whatsoever.

Elbandi, 2020-08-10:
i rebuild the same image without encrypt feature, and it's 'same' error: https://imgur.com/a/TgsTy1I
(=it could not find the backend device). there is a supported bananapi M1 hw, if i checked goo, it's using same source tree, so _maybe_ affected too. but i cannot test, i havent m1 :(

i'm stepping back to source commits, and trying the builds...

igorpecovnik, 2020-08-10:
> i rebuild the same image without encrypt feature

There is no support for this hardware.
https://github.com/armbian/build#support

closed 2020-08-10 by igorpecovnik