armbian/build #2146: Analog Codec with different versions

FRBOUCHER, 2020-08-14:
Hello,

Depending on the versions of the system used, I found a difference.
I have used an older version Armbian_5.91_Orangepizero_Debian_buster_next_4.19.59.7z and the latest version Armbian_20.05.3_Orangepizero_buster_current_5.4.45.img.xz.
With armbian-config in System then Hardware, if I set the analog codec, I get two results for / proc / asound / modules: with V4.19.59: (null) - and - with V5.4.45: (efault).
I wonder about these results. Is this normal? Do you have to do another action first to get the same result? Can we run a specific command to define this codec without going through the configuration utility? Can it influence the GPIO management mechanism?

With thanks
François Boucher

igorpecovnik, 2020-08-14:
Please use forum for technical support - remember that searching forum saves developers precious private time. This is not a commercial product.

1. Stay here for a long time: http://armbian.com/search
2. If you can't find, then you ask question here: https://forum.armbian.com/forum/13-allwinner-h2-h3/

closed 2020-08-14 by igorpecovnik

FRBOUCHER, 2020-08-20:
Please tell me where I can post my message. I am frustated by your action to close it.
Thank's
François Boucher

EvilOlaf, 2020-08-20:
Since you are using an OrangePi Zero consider using the **Allwinner H2 & H3** forums. Your first post may need to be approved. This sadly is needed due to massive spam bots all over the place.

The issue tracker on this repository is dealing with issues in the build process only. Everything beyond is discussed in forums.

igorpecovnik, 2020-08-20:
>  I am frustated

I am sorry, but on demand support is far too expensive that we can finance with our private cash for you. Best effort forum support or hiring 3rd party expert is what your "i want support and attention for free" contract covers: https://github.com/armbian/build#support

I am seeing this every day. Now who is more frustrated?