armbian/build #2147: Official support for Banana Pi R2

martinl, 2020-08-14:
I am interested in working on making Banana Pi R2 support from current state to officially supported in Armbian. Can you please explain what are the requirements for official support and what is currently missing for Banana Pi R2 

I am currently running one R2 on minimal Debian buster image prebuilt by https://github.com/frank-w/ with latest kernel from https://github.com/frank-w/BPI-R2-4.14 5.4-main branch. 

It has been stable for about 2 weeks in normal use, running some lxc containers. I have not tested onboard wifi and BT yet but booting both from SD card and eMMC works. Also USB and wired network seem to be working fine so far. HDMI does work but has minor sync/frequency issues with my 720p pico projector.

with best regards,
Martin

igorpecovnik, 2020-08-14:
> Can you please explain what are the requirements for official support and what is currently missing for Banana Pi R2

If you can maintain it, test on upgrades, support users on forum and if [you find a friend](https://forum.armbian.com/topic/7296-bananapi-r2-csc-mt7623-as-new-boardfamily/). If you succeeds with minimum requirements and you two(!) stays around, I will put my R2 board to the [automated test rig](https://twitter.com/armbian/status/1257639172778622977) that we will monitor basic functions on upgrades and change status. Current crew is way too small to add this board on the top of existing - in case you (two) decides to not do this anymore ... things go back.

martinl, 2020-08-15:
Thanks for explaning the requirements. I'll set up the build environment for testing and see what I can do about support part

martinl, 2020-08-15:
I built and flashed the legacy version for R2 and the autotests are currently running from SD:
[image]

[image]

in general, what would I have to do to add R2 support for 5.4 kernel (using frank-w 5.4-main patches)

igorpecovnik, 2020-08-15:
> what would I have to do to add R2 support for 5.4 kernel

https://github.com/armbian/build#support

closed 2020-08-15 by igorpecovnik

igorpecovnik, 2020-08-17:
https://github.com/armbian/build/issues/1818

martinl, 2020-08-17:
thanks, I was looking for that issue thread yesterday

martinl, 2020-08-18:
My current R2 porting efforts progress is at https://github.com/martinl/banana-pi-r2/ and I'll push tested patches here https://github.com/martinl/build