armbian/build #2150: locale

AaronNGray, 2020-08-19:
I am getting the following in the build even though I have installed locale's as below and used expert mode and set the locale.

```
sudo apt install locales-all && sudo locale-gen
```
```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "en_US:en",
        LC_ALL = "en_US.UTF-8",
        LANG = (unset)
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
```

igorpecovnik, 2020-08-28:
On a clean host we don't see this problem.

closed 2020-08-29 by AaronNGray

Legogris, 2021-02-01:
@igorpecovnik I'm actually getting this now on a fresh vagrant box on current master:

```
$ ./compile.sh DEST_LANG=en_US.UTF-8 BOOTSIZE=500 BOARD=odroidhc4 BRANCH=current BUILD_DESKTOP=no RELEASE=bullseye EXPERT=yes CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE="xxxxx" PROGRESS_DISPLAY=plain KERNEL_ONLY=no CLEAN_LEVEL=images CRYPTROOT_SSH_UNLOCK_PORT=2222 CRYPTROOT_SSH_UNLOCK=yes

[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ focal ]
[ o.k. ] Updating from external repository [ aptly ]
[ .... ] Installing build dependencies
[...]
I: Extracting liblzma5...
I: Extracting zlib1g...
[ o.k. ] Installing base system [ Stage 2/2 ]
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
W: Failure trying to run:  /sbin/ldconfig
W: See //debootstrap/debootstrap.log for details
[ error ] ERROR in function create_rootfs_cache [ debootstrap.sh:177 ]
[ error ] Debootstrap base system second stage failed
[ o.k. ] Process terminated
[ o.k. ] Unmounting [ /home/vagrant/armbian/.tmp/rootfs-current-odroidhc4-bullseye-no-no/ ]
Device armbian-root is not active.
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:66 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

It was working fine the other day (after which I did halt/destroy/upgrade/up again)

igorpecovnik, 2021-02-01:
Not a problem of locale but qemu / debootstrap bug. Also BULLSEYE is not supported target.