armbian/build #2157: Rockchip64 kernel compilation fail

gleam2003, 2020-08-25:
Kernel compilation for Rockchip64 family fail.
The problem is the patch file "general-bluetooth-add-new-quirk.patch"

Tonymac32, 2020-08-25:
Which kernel version were you compiling? Does it succeed if you remove the
patch?

piter75, 2020-08-25:
@gleam2003 Beside providing the kernel version please also verify that you are using the latest master for building.

The issue with latest kernel patch releases should be fixed already in both current (https://github.com/armbian/build/commit/403651b986c365ee60eb282c1e5c256d1c4e3a0b) and dev (https://github.com/armbian/build/commit/a5018382c4748930196102f591539faeac337c40).

gleam2003, 2020-08-25:
Sorry, my error.
I put .ignore_changes and don't see te update.

closed 2020-08-25 by gleam2003