armbian/build #2159: Missing toolchains. Can't build anything

xlazom00, 2020-08-25:
since last commit @igorpecovnik  https://github.com/armbian/build/commit/2c3a15f2d94df6f46d404ac085f9af66c09d0b56
we can't build any image
```general.sh: line 1191: cd: build/cache/toolchain: No such file or directory```

umiddelb, 2020-08-25:
workaround: `mkdir cache/toolchain` after 1st try

umiddelb, 2020-08-25:
There seems to be a [mismatch bug](https://github.com/umiddelb/build/commit/d1dcea44f59ddc42325748081b0e72da56038359) in `./lib/compilation.sh` and/or anywhere else. The toolchain archives are extracted in `./cache/toolchain` but the directory isn't created automatically. The check function refers to `./cache/toolchains` which exists but remains empty.

igorpecovnik, 2020-08-25:
I had to change the location due to ... very long story. Is this a fix now?

umiddelb, 2020-08-25:
There is at least one open bug. The script which creates ./cache/toolchains should create ./cache/toolchain instead.

igorpecovnik, 2020-08-25:
https://github.com/armbian/build/pull/2160

umiddelb, 2020-08-25:
it's difficult to test, since ./compile.sh switches back to the master branch automatically.

closed 2020-08-25 by igorpecovnik