armbian/build #2185: armbian-check-first-login.sh

matt335672, 2020-09-04:
Hi,

I'm one of the maintainers of the XRDP project.

We've identified an issue [#1672](https://github.com/neutrinolabs/xrdp/issues/1672) which happens on Armbian when the Debian xrdp RPM is installed. Essentially the xrdp RPM is packaged with a script which sources `/etc/profile` from `/bin/sh`. This blows up on Armbian, as `/etc/profile.d/armbian-check-first-login.sh` contains quite a few bash-isms which the default shell linked from `/bin/sh` doesn't like. 

You can demonstrate this on the Armbian command line (no XRDP needed) with:-

```
sh -x /etc/profile
. . .
/etc/profile: 20: /etc/profile.d/armbian-check-first-login.sh: Syntax error: "(" unexpected
```

To my mind, scripts in `/etc/profile.d` should be source-able by any Bourne-compatible shell on the system, as any one of these could be used as a user's login shell.

I could see three ways to make this script compatible with `/bin/sh`:- 
1. Remove the bash-isms.
2. Add a header to the file to make sure bash is used for the script (e.g.):-
   ```bash
   # Make sure this script is executed with bash when called from
   # /etc /profile.d
   if [ -z "$BASH" ]; then
       if [ -z "-" ]; then
           exec /bin/bash $0 "$@"
       else
           exec /bin/bash -$- $0 "$@"
       fi
       exit $?
   fi
   ```
3. Move the body of the script elsewhere and replace `/etc/profile.d/armbian-check-first-login.sh` with a simple script to call it.

I'm happy to put a PR together for any of the above, but I don't know which (if any) you'd prefer. My own thoughts are:-
- Option 1 is certainly possible but there are quite a few bash-isms in the script, and there would be no guarantees that more couldn't creep back in later.
- Option 2 seems to work with the basic testing I've done so far - I can do more.
- Option 3 is cleaner than option 2 to my mind, but you may have good reasons not to do this.

Thanks for any info

lanefu, 2020-09-05:
Hi @matt335672 

Thanks for identifying the issue.  It does make sense to not assume bash.   Thank you for offering a PR.

I like the idea of moving the scripts and simply calling them.   @igorpecovnik any concerns?

A PR for Option 3 would mainly consist of moving / refactoring the contents [here](https://github.com/armbian/build/tree/master/packages/bsp/common/etc/profile.d) and putting them with the other armbian libs [here](https://github.com/armbian/build/tree/master/packages/bsp/common/usr/lib/armbian)

igorpecovnik, 2020-09-05:
> any concerns?

Moving to our common lib and make a link? No concerns.

Due to some other bug in BSP scripts, I plan to push out an update just for bsp packages in the following days.

matt335672, 2020-09-05:
Hi @igorpecovnik 

When you say you're planning on pushing out an update, will you be addressing this issue or would you still like me to contribute? Happy either way.

igorpecovnik, 2020-09-05:
This issue is important to be fixed and I will wait. If you are able to deal with this, you are welcome!

closed 2020-09-07 by igorpecovnik

lanefu, 2020-09-07:
thank you @matt335672

PR #2190 (2020-09-07): Move armbian-check-first-login.sh to separate bash script
Resolves #2185 

It's not easy to see exactly what is going on from the change log below, but essentially:-
- `armbian-check-first-login.sh`is replaced with a minimal sh-compatible script which checks for root's first login and calls `/usr/lib/armbian/armbian-firstlogin` if it detects this.
- `/usr/lib/armbian/armbian-firstlogin` is exactly the same as `armbian-check-first-login.sh` used to be, but with this patch applied:-
```diff
-#!/bin/sh
+#!/bin/bash
 #
 # Copyright (c) Authors: http://www.armbian.com/authors
 #
@@ -183,7 +183,7 @@ [... cut]