armbian/build #2189: Armbian-config missing in latest build

g-provost, 2020-09-07:
In latest build 20.08.1  (build on 04/09), armbian-config tool is missing.

Checked in Helios64 Buster and Focal (kernel 5.8.6).

Can someone confirm if same issue on other targets ?

igorpecovnik, 2020-09-07:
In normal builds, right? Minimal comes without.

g-provost, 2020-09-07:
Humm so now all the image build are by default minimal build ?  Previous images were still normal, since armbian-config was present.

How to know an image is minimal based on its filename ?

g-provost, 2020-09-07:
Sorry I misread your question. Yes armbian-config is missing in normal build.

igorpecovnik, 2020-09-07:
And it looks only when building all images. Single image - checked focal and bullseye, both has it.

g-provost, 2020-09-09:
So do you mean there is an issue in certain build condition ?

igorpecovnik, 2020-09-09:
> So do you mean there is an issue in certain build condition ?

We have some issues with "build-all" which is done simultaneously. There is possible a race condition involved but I am not sure if this is related.

My first quick test couldn't reproduce this problem. I also rebuild Helios64 images few days ago but I haven't specifically check if armbian-config is missing or not.

igorpecovnik, 2020-10-06:
This was not possible to reproduce. It was probably a network issues.

closed 2020-10-06 by igorpecovnik