armbian/build #2207: Sound for all AML (N2 etc)

150balbes, 2020-09-14:
This is a proven and working feature that ensures that HDMI audio works on all AML devices with a core >=5.7.
These elements ensure that the sound is turned on for the first time on all AML devices (including sound  RK and AW).

https://github.com/150balbes/Build-Armbian/blob/master/config/sources/families/arm-64.conf#L163-L164
https://github.com/150balbes/Build-Armbian/blob/master/packages/bsp/arm-64/sound.sh
https://github.com/150balbes/Build-Armbian/tree/master/packages/bsp/arm-64/cards
https://github.com/150balbes/Build-Armbian/blob/master/packages/bsp/common/usr/lib/armbian/armbian-firstlogin#L261

wiresharkkkk, 2020-10-03:
Hello..!
I've been searching for an issue where the audio is not connected in the hdmi and I'm here.

`apt-get -y install git
git clone https://github.com/150balbes/Build-Armbian
cd Build-Armbian
./compile.sh`

I entered the above command and rebooted, but there is still no audio output.

Only dummy is output in my audio.

Have I ever missed something?

igorpecovnik, 2020-10-03:
> Have I ever missed something?

Yes. You missed support terms https://github.com/armbian/build#support

closed 2020-10-18 by igorpecovnik