armbian/build #2215: rockchip64 - patch failing to apply

EvilOlaf, 2020-09-18:
```
Processing file /home/build2/build/patch/kernel/rockchip64-current/wifi-4003-fix-sha256_state-clashes.patch
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/net/wireless/rtl8723ds/include/rtw_security.h.rej
Reversed (or previously applied) patch detected!  Skipping patch.
5 out of 5 hunks ignored -- saving rejects to file drivers/net/wireless/rtl8723ds/core/rtw_security.c.rej
```
Maybe deprecated? Did not dive deeper.

igorpecovnik, 2020-09-18:
Did you update?
https://github.com/armbian/build/commit/ef50295b327eb1e2e199ac54e7edd92e7f003d9c

EvilOlaf, 2020-09-18:
Yeah, I usually do a pull before doing messing around.
Well I assume my pull must have happened just a hair before the mentioned push.
Everything is good.

closed 2020-09-18 by EvilOlaf