armbian/build #2223: Build fail v20.08 and V20.05

ludiazv, 2020-09-25:
Hi,
   Trying to build new stable using the build system. I use standard vargrant build method with the minor changes of increasing the CPUs and memory of the VM to speed up the build.

  I tried last stable by checking out the last tag in this repo (v20.08) and launched a build for the opi zero. 
   The build ran as usual and uboot compiled successfully but the kernel failed:
```
  CC [M]  fs/xfs/xfs_sysctl.o
  CC [M]  fs/xfs/xfs_pnfs.o
  LD [M]  fs/xfs/xfs.o
[ error ] ERROR in function compile_kernel [ compilation.sh:411 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 
```

Checking the compilation logs I discovered:

compilation.log
````
	== u-boot ==

.config:1499:warning: override: reassigning to symbol BOOTDELAY
arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb: Warning (reg_format): /soc/ethernet@1c30000/ethernet-phy@1:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb: Warning (avoid_default_addr_size): /soc/ethernet@1c30000/ethernet-phy@1: Relying on default #address-cells value
arch/arm/dts/sun8i-h3-orangepi-zero-plus2.dtb: Warning (avoid_default_addr_size): /soc/ethernet@1c30000/ethernet-phy@1: Relying on default #size-cells value
board/sunxi/board.c: In function ‘ft_board_setup’:
board/sunxi/board.c:957:2: warning: implicit declaration of function ‘fdt_fixup_ethernet’ [-Wimplicit-function-declaration]
  957 |  fdt_fixup_ethernet(blob);
      |  ^~~~~~~~~~~~~~~~~~

	== kernel ==

drivers/video/backlight/pwm_bl.c: In function ‘pwm_backlight_probe’:
drivers/video/backlight/pwm_bl.c:643:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  643 |   u64 level = (u64)state.duty_cycle * pb->scale;
      |   ^~~
In file included from ./include/linux/printk.h:7,
                 from ./include/linux/kernel.h:15,
                 from ./include/asm-generic/bug.h:19,
                 from ./arch/arm/include/asm/bug.h:60,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/thread_info.h:12,
                 from ./include/asm-generic/current.h:5,
                 from ./arch/arm/include/generated/asm/current.h:1,
                 from ./include/linux/sched.h:12,
                 from ./include/linux/ptrace.h:6,
                 from ./arch/arm/include/asm/kgdb.h:14,
                 from ./include/linux/kgdb.h:20,
                 from ./include/linux/fb.h:5,
                 from ./include/linux/bootsplash.h:13,
                 from drivers/video/fbdev/core/bootsplash_load.c:15:
drivers/video/fbdev/core/bootsplash_load.c: In function ‘bootsplash_load_firmware’:
./include/linux/kern_levels.h:5:18: warning: format ‘%li’ expects argument of type ‘long int’, but argument 2 has type ‘size_t’ {aka ‘const unsigned int’} [-Wformat=]
    5 | #define KERN_SOH "\001"  /* ASCII Start Of Header */
      |                  ^~~~~~
./include/linux/kern_levels.h:14:19: note: in expansion of macro ‘KERN_SOH’
   14 | #define KERN_INFO KERN_SOH "6" /* informational */
      |                   ^~~~~~~~
./include/linux/printk.h:368:9: note: in expansion of macro ‘KERN_INFO’
  368 |  printk(KERN_INFO pr_fmt(fmt), ##__VA_ARGS__)
      |         ^~~~~~~~~
drivers/video/fbdev/core/bootsplash_load.c:95:2: note: in expansion of macro ‘pr_info’
   95 |  pr_info("Loading splash file (%li bytes)\n", fw->size);
      |  ^~~~~~~
drivers/video/fbdev/core/bootsplash_load.c:95:34: note: format string is defined here
   95 |  pr_info("Loading splash file (%li bytes)\n", fw->size);
      |                                ~~^
      |                                  |
      |                                  long int
      |                                %i
In file included from ./include/linux/printk.h:7,
[... 175 lines cut ...]
Displaying message: Installing build dependencies
Displaying message: Syncing clock host info
Displaying message: Checking for external GCC compilers  info
gpg: keybox '/home/vagrant/armbian/cache/.gpg/pubring.kbx' created
gpg: /home/vagrant/armbian/cache/.gpg/trustdb.gpg: trustdb created
gpg: error reading key: No public key
gpg: key D47877808F427EAF: public key "Linaro Toolchain Builder <michael.hope+cbuild@linaro.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: error reading key: No public key
gpg: key 93D6889F9F0E78D5: 2 signatures not checked due to missing keys
gpg: key 93D6889F9F0E78D5: public key "Igor Pecovnik <igor@armbian.com>" imported
gpg: no ultimately trusted keys found
gpg: Total number processed: 1
gpg:               imported: 1
gpg: no signed data
gpg: can't hash datafile: No data
Displaying message: ERROR in function download_and_verify general.sh:1177 err
Displaying message: verification failed  err
Displaying message: Process terminated  info

````
How can this be fixed?

Thanks,

igorpecovnik, 2020-09-25:
> How can this be fixed?

Use master. We haven't pull those fixes to older branches.

igorpecovnik, 2020-09-25:
v20.08 should be fixed.

ludiazv, 2020-09-25:
Hi @igorpecovnik ,
   The first error is with V20.08 tag. And do not compile as you can see in the first two traces.
 
   For V20-05 is a key problem. I'm also interested to solve it but if not possible no problem.

I will check with current master. But I think V20.08 as it declared stable should compile.

Many thanks.

igorpecovnik, 2020-09-25:
> But I think V20.08 as it declared stable should compile.

After this https://github.com/armbian/build/commit/73ae8fe060b8f4ec3f51afb3a5117aa50dbf60f5 it should. But I didn't have time to test.

ludiazv, 2020-09-26:
I finally got it.
  The problem was that I was checking out **'v20.08' tag** of the master branch not the *'v20.08' branch*. I think is not a good idea to have tags on master with same name that a branch to avoid naming collisions. I think this can be avoided in future releases.

 If you do they typical ``git checkout v20.08`` it checks our the tag by default not the branch. To check out the branch you must find the ref and check it out with the full name.

Thanks for the help.

closed 2020-09-26 by ludiazv

NicolasDorier, 2020-10-10:
@igorpecovnik got this issue, the fact that the tags and the branch share same name but not same commit is prone to foot shooting.

igorpecovnik, 2020-10-10:
>  is prone to foot shooting.

I agree its problematic. Its a problem we created, but it affects little people so the priority for fixing is low ...  project maintaining resources are extremely limited, best case scenario - if we receive no help - is easily one year from now.

My Gitfu is too low to properly address this mess which probably needs some better organizing at this scale. Someone - we don't have - have to do that ...

NicolasDorier, 2020-10-13:
This should do it. (untested)
```bash
git tag -d v20.08
git push -d origin tags/v20.08
git checkout v20.08
git tag -a v20.08 -m v20.08
git push origin v20.08
```

igorpecovnik, 2020-10-13:
Automatic way of merging is not a problem - but we don't want that. We only want to keep bugfixes in, new features out. Now - can you script it? :)