armbian/build #2245: rockchip64-current issues with CONFIG_STACKPROTECTOR_PER_TASK

jbergler, 2020-10-08:
In https://github.com/armbian/build/commit/a032106fc074d9eaf42074266209bc9e71884965 config/kernel/linux-rockchip64-current.config was updated to gcc 9.

As a result the following options were enabled 

```
CONFIG_CC_HAVE_STACKPROTECTOR_SYSREG=y
CONFIG_STACKPROTECTOR_PER_TASK=y
```

This breaks compiling kernel modules on buster since gcc 8 does not support the compiler options required
```
$ gcc --version
gcc (Debian 8.3.0-6) 8.3.0

$ gcc -mstack-protector-guard=sysreg -mstack-protector-guard-reg=sp_el0 -mstack-protector-guard-offset=0
gcc: error: unrecognized command line option '-mstack-protector-guard=sysreg'; did you mean '-fstack-protector-strong'?
gcc: error: unrecognized command line option '-mstack-protector-guard-reg=sp_el0'; did you mean '-fstack-protector-all'?
gcc: error: unrecognized command line option '-mstack-protector-guard-offset=0'; did you mean '-fstack-protector-strong'?
gcc: fatal error: no input files
compilation terminated.
```

The symptoms were reported in some other projects:
- https://github.com/google-coral/edgetpu/issues/228
- https://github.com/openzfs/zfs/issues/10985

Is it possible to disable this feature?

jbergler, 2020-10-10:
It appears this issue also exists on current-meson64

zhangn1985, 2020-10-10:
why you are using host gcc? or you are using native arm64 gcc?

jbergler, 2020-10-10:
This is native arm64 gcc trying to compile kernel modules

igorpecovnik, 2020-11-30:
I checked a bit but didn't dig into a kernel much. Proposed fix doesn't work.

Is upgrading GCC on Buster viable workaround? Or might bring many other problems? In any case it is some work, while we don't have resources to just jump on this and fix it.

jbergler, 2020-12-02:
If we built the kernel on buster with the default tool chain the per task stack protection would be off anyway.
The kernel flag is macrod to include a compiler check for the feature.

If we backported a newer tool chain it would likely work, but that sounds like an unmaintainable solution.

My feeling is that we should disable this in 5.8+ kernels that ship for platforms with an older tool chain until they’re gone.

It’s a small price to pay for folks on the focal images but saves the hassle of having separate kernels or backporting gcc

closed 2021-01-27 by igorpecovnik