armbian/build #2246: rockchip (not rockchip64) DTBs are not built with -@

nwf, 2020-10-10:
I believe this is the right place to report this; if not, please point me in the right direction.

I have a tinkerboard running `linux-image-current-rockchip` and would like to avail myself of the `user_overlays` feature of `/boot/armbianEnv.txt`.  Everything works fine except that `rk3288-tinker.dtb` wasn't built with `-@` and so lacks the `__symbols__` necessary (yes?) for overlays to work.  If I manually do that and pave over the file in `/boot/dtb`, everything works fine.

https://github.com/armbian/build/blob/master/patch/kernel/rockchip64-current/general-add-overlay-compilation-support.patch seems to be adding `-@` to `DTC_FLAGS` for `rockchip64` and so perhaps a similar patch in `patch/kernel/rockchip-current` would suffice.

Tonymac32, 2020-10-18:
https://github.com/armbian/build/commit/4cdf3c3d0e31fb9fa05d9b818915d54c465dffa0

If you can build an image see if this gave you the capability you're looking for, it's a general patch and should work.

paolosabatino, 2021-03-16:
I close the issue because the `-@` looks to be there for both rockchip-current and rockchip-dev:
https://github.com/armbian/build/blob/06a864a39c7e298eacf2d9b90da3b3a74f083ea1/patch/kernel/rockchip-current/general-add-overlay-compilation-support.patch#L54
https://github.com/armbian/build/blob/06a864a39c7e298eacf2d9b90da3b3a74f083ea1/patch/kernel/rockchip-dev/general-add-overlay-compilation-support.patch#L54

closed 2021-03-16 by paolosabatino