armbian/build #2248: Failure to build kernel on v20.08:  error: ‘cfg80211_rtw_mgmt_frame_register’ undeclared here

NicolasDorier, 2020-10-10:
Based on branch v20.08, I do not manage to build the kernel:

```
drivers/net/wireless/rtl8189es/os_dep/linux/ioctl_cfg80211.c:6549:37: error: ‘cfg80211_rtw_mgmt_frame_register’ undeclared here (not in a function)
 6549 |  .update_mgmt_frame_registrations = cfg80211_rtw_mgmt_frame_register,
      |                                     ^~~~~~~~~~~
```

Config
```
[ o.k. ] Running the container
[ o.k. ] Using config file [ /root/armbian/userpatches/config-example.conf ]
[ o.k. ] Command line: setting BOARD to [ rockpro64 ]
[ o.k. ] Command line: setting KERNEL_ONLY to [ no ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ no ]
[ o.k. ] Command line: setting RELEASE to [ focal ]
[ o.k. ] Command line: setting BRANCH to [ current ]
[ o.k. ] Command line: setting BUILD_DESKTOP to [ no ]
[ o.k. ] Command line: setting WIREGUARD to [ no ]
[ o.k. ] Command line: setting BUILD_MINIMAL to [ yes ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ focal ]
[ o.k. ] Running in container [ docker ]
....
[ o.k. ] Building deb [ linux-u-boot-current-rockpro64_20.08.4_arm64.deb ]
[ o.k. ] Cleaning [ linux-mainline/linux-5.8.y ]
[ o.k. ] Adjusting [ packaging ]
[ o.k. ] * applying general-packaging-5.6.y.patch
[ o.k. ] Adding [ Kernel splash file ]
[ o.k. ] * applying bootsplash-5.8.10-0001-Revert-vgacon-remove-software-scrollback-support.patch
[ o.k. ] * applying bootsplash-5.8.10-0002-Revert-fbcon-remove-now-unusued-softback_lines-curso.patch
[ o.k. ] * applying bootsplash-5.8.10-0003-Revert-fbcon-remove-soft-scrollback-code.patch
[ o.k. ] * applying 0001-bootsplash.patch
[ o.k. ] * applying 0002-bootsplash.patch
[ o.k. ] * applying 0003-bootsplash.patch
[ o.k. ] * applying 0004-bootsplash.patch
[ o.k. ] * applying 0005-bootsplash.patch
[ o.k. ] * applying 0006-bootsplash.patch
[ o.k. ] * applying 0007-bootsplash.patch
[ o.k. ] * applying 0008-bootsplash.patch
[ o.k. ] * applying 0009-bootsplash.patch
[ o.k. ] * applying 0010-bootsplash.patch
[ o.k. ] * applying 0011-bootsplash.patch
[ o.k. ] * applying 0012-bootsplash.patch
[ o.k. ] Adding [ Wireless package injections for mac80211 compatible chipsets ]
[ o.k. ] * applying kali-wifi-injection-1-pre-v5.9.patch
[ o.k. ] * applying kali-wifi-injection-2.patch
[ o.k. ] * applying kali-wifi-injection-3.patch
[ o.k. ] Adding [ Drivers for 2.5Gb RTL8152/RTL8153 USB dongles branch:master ]
[ o.k. ] Checking git sources [ rtl8152 master ]
[ .... ] Up to date
[ o.k. ] Adding [ Wireless drivers for Realtek 8189ES chipsets branch:master ]
[ o.k. ] Checking git sources [ rtl8189es master ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Adding [ Wireless drivers for Realtek 8189FS chipsets branch:rtl8189fs ]
[ o.k. ] Checking git sources [ rtl8189fs rtl8189fs ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Adding [ Wireless drivers for Realtek 8192EU chipsets branch:realtek-4.4.x ]
[ o.k. ] Checking git sources [ rtl8192eu realtek-4.4.x ]
[ .... ] Up to date
[ o.k. ] Adding [ Wireless drivers for Realtek 8811, 8812, 8814 and 8821 chipsets branch:v5.6.4.2 ]
[ o.k. ] Checking git sources [ rtl8812au v5.6.4.2 ]
[ .... ] Up to date
[ o.k. ] Adding [ Wireless drivers for Realtek RTL8811CU and RTL8821C chipsets commit:2bebdb9a35c1d9b6e6a928e371fa39d5fcec8a62 ]
[ o.k. ] Checking git sources [ rtl8811cu 2bebdb9a35c1d9b6e6a928e371fa39d5fcec8a62 ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Adding [ Wireless drivers for Realtek 8188EU 8188EUS and 8188ETV chipsets branch:v5.7.6.1 ]
[ o.k. ] Checking git sources [ rtl8188eu v5.7.6.1 ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] * applying wireless-rtl8188eu.patch
[ o.k. ] Adding [ Wireless drivers for Realtek 88x2bu chipsets branch:5.6.1_30362.20181109_COEX20180928-6a6a ]
[ o.k. ] Checking git sources [ rtl88x2bu 5.6.1_30362.20181109_COEX20180928-6a6a ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Adding [ Wireless drivers for Realtek 8723DS chipsets branch:master ]
[ o.k. ] Checking git sources [ rtl8723ds master ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Adding [ Wireless drivers for Realtek 8723DU chipsets branch:master ]
[ o.k. ] Checking git sources [ rtl8723du master ]
[ .... ] Up to date
[ o.k. ] * applying wireless-rtl8723du.patch
[ o.k. ] Started patching process for [ kernel rockchip64-current ]
[ o.k. ] Looking for user patches in [ userpatches/kernel/rockchip64-current ]
```

```
[ error ] ERROR in function compile_kernel [ compilation.sh:411 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
```

Logs
```
drivers/net/wireless/rtl8189es/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
drivers/net/wireless/rtl8189es/core/rtw_mlme_ext.c:1093:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1093 |    if(check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
      |      ^
drivers/net/wireless/rtl8189es/core/rtw_mlme_ext.c:1098:3: note: here
 1098 |   case WIFI_ASSOCREQ:
      |   ^~~~
drivers/net/wireless/rtl8189es/core/rtw_iol.c: In function ‘rtw_IOL_cmd_buf_dump’:
drivers/net/wireless/rtl8189es/core/rtw_iol.c:261:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  261 |   if(j%32 ==0) printk("\n");j++;
      |   ^~
drivers/net/wireless/rtl8189es/core/rtw_iol.c:261:29: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  261 |   if(j%32 ==0) printk("\n");j++;
      |                             ^
drivers/net/wireless/rtl8189es/os_dep/linux/ioctl_cfg80211.c:6549:37: error: ‘cfg80211_rtw_mgmt_frame_register’ undeclared here (not in a function)
 6549 |  .update_mgmt_frame_registrations = cfg80211_rtw_mgmt_frame_register,
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [scripts/Makefile.build:281: drivers/net/wireless/rtl8189es/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** [scripts/Makefile.build:497: drivers/net/wireless/rtl8189es] Error 2
make[2]: *** [scripts/Makefile.build:497: drivers/net/wireless] Error 2
make[1]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make: *** [Makefile:1752: drivers] Error 2
```

igorpecovnik, 2020-10-10:
Use master branch or step us as a day-day maintaner of that branch that will manually pull bugfixes into that branch and keep it up to date. Option B - that we will do that for you - is not going to happen.

closed 2020-10-10 by igorpecovnik

NicolasDorier, 2020-10-10:
@igorpecovnik I may have not understood correctly, but I thought that 20.08 was a "release". If I find the correct commit fixing this error, should I make a backport to 20.08, or you will just make a new release later?

igorpecovnik, 2020-10-10:
> If I find the correct commit fixing this error, should I make a backport to 20.08 and make PR for it, or you will just make a new release later?

Only if you plan of doing this on a long term, otherwise, for one time fix, don't bother. In a few weeks, some other (major) problem might pop up ... it is possible to fix this problem to freeze (unfreez on bugfix) commits on all sources. There are a lot of them and represent a serious maintainace issue and I can assure you that we will not do anything about. We can't. 

Those releases are in fact only used for building images - we branch once we are ready https://docs.armbian.com/Process_Release-Model/#release-branching-versioning-and-tags to release master for further development. But we only keep last stable branch maintained for about 1 month to issue smaller bug fix releases, then we move on. This is the best possible and we will no imprve this with current crew - also make sure to check support terms https://github.com/armbian/build#support

NicolasDorier, 2020-10-11:
If somebody get stuck, cherry-picking this commit on top of the branch fixed my issue:

https://github.com/armbian/build/commit/f28912f041985e6341c5efcaa6218234738a53d2