armbian/build #2249: ODroid-N2 armbian-config hardware menu not work

iav, 2020-10-10:
current ODroid-N2 focal image have not working system→hardware menu https://snipboard.io/KY1TxC.jpg

The reason is missing of `/boot/armbianEnv.txt` file.
After file creation with 2 variables:
```
cat /boot/armbianEnv.txt 
overlays=amlogic 
overlay_prefix=meson
```
configuration start to work: https://snipboard.io/4rj65f.jpg

EvilOlaf, 2020-10-11:
Please report to this repo: https://github.com/armbian/config

igorpecovnik, 2020-10-11:
The problem is related to the bootscripts / env creation - which is a part of the build script engine. Once fixed, it will work in armbian-config.

iav, 2020-10-11:
I found this fragment https://github.com/armbian/build/blob/e395961f77b1b38f85507806ffde9e2693f42cca/lib/makeboarddeb.sh#L210
Why it not work for N2?

igorpecovnik, 2020-10-11:
N2 has boot.ini which is plain text format and is designed to be directly editable. Boot.cmd - which is default u-boot way - needs compilation to binary boot.scr which is actually loaded. Here boot.ini is loaded directly. Well, some RFC or moving to mainline u-boot is needed to get this right. None is straightforward.

iav, 2020-10-12:
I trace building process and found this file present during first image creation:
armbian/.tmp/rootfs-current-odroidn2-focal-no-no/boot/armbianEnv.txt
Then it dasappears.
Trying to find when and why.

iav, 2020-10-12:
during build:
```
cat /home/xxx/armbian/.tmp/rootfs-current-odroidn2-focal-no-no/boot/armbianEnv.txt
verbosity=1
console=both
ethaddr=00:50:43:84:fb:2f
overlay_prefix=meson
```

igorpecovnik, 2020-10-12:
https://github.com/armbian/build/blob/master/lib/debootstrap.sh#L512-L522

iav, 2020-10-12:
But why?
Why line 521 should exist? Is it possible to remove it?
Total, or, at least, for meson family?

igorpecovnik, 2020-10-12:
Don't remember. You can make it work for meson64 family ...

iav, 2020-10-12:
How can I test upgrade? Setup release to sdcard, then...? 
Install with dpkg freshest debs I got in `output/debs` and `output/debs/release`?

igorpecovnik, 2020-10-12:
install fresh package with dpkg should do yes.

iav, 2020-10-13:
I did test upgrade — works for me.

closed 2020-10-13 by igorpecovnik