armbian/build #2250: Build for legacy kernel RockPI-4c don't work

britus, 2020-10-11:
Hi,
by choosing Old stable / Legacy see attachment!
/root/projects/armbian/build/lib/main.sh: line 516: eval: -h: invalid option
eval: usage: eval [arg ...]

igorpecovnik, 2020-10-11:
Works here without any problem.

Without debug logs and so little vague information its difficult to proceed. What is your host or virtualization type?

britus, 2020-10-12:
Hi Igor,
well, it's simply '/root/projects/armbian/build/lib/main.sh: line 516: eval: **-h: invalid option**
I starting the compile.sh script with command line option '-h' ;)
... and I know ... the 'f$c$k' user are some time so stupid :) blame me.

closed 2020-10-12 by britus

igorpecovnik, 2020-10-12:
Ahaa :)

-help is planned but not started yet.
https://armbian.atlassian.net/browse/AR-375