armbian/build #2259: Debian build fails at last (?) step with exec format error

voltagex, 2020-10-18:
The kernel built successfully, something else is failing here

Sidenote: can I please get an option to disable the use of bittorrent?

```
dpkg-deb: building package 'linux-source-5.8.16-current-rockchip64' in '/home/voltagex/src/armbian-build/.tmp/linux-source-current-rockchip64_20.11.0-trunk_all.deb'.
[ o.k. ] Building deb [ armbian-config ]
[ o.k. ] Checking git sources [ armbian-config master ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 23 (delta 0), reused 10 (delta 0), pack-reused 0
Unpacking objects: 100% (23/23), done.
From https://github.com/armbian/config
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Enumerating objects: 269, done.
remote: Counting objects: 100% (269/269), done.
remote: Compressing objects: 100% (216/216), done.
remote: Total 269 (delta 40), reused 261 (delta 39), pack-reused 0
Receiving objects: 100% (269/269), 10.90 MiB | 32.65 MiB/s, done.
Resolving deltas: 100% (40/40), done.
From https://github.com/armbian/firmware
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Up to date
[ o.k. ] Checking git sources [ linux-firmware-git master ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Enumerating objects: 2210, done.
remote: Counting objects: 100% (2210/2210), done.
remote: Compressing objects: 100% (1539/1539), done.
remote: Total 2210 (delta 800), reused 1809 (delta 618), pack-reused 0
Receiving objects: 100% (2210/2210), 186.78 MiB | 30.99 MiB/s, done.
Resolving deltas: 100% (800/800), done.
From https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
[ o.k. ] Creating board support package [ rockpro64 current ]
[ .... ] Fingerprinting
[ o.k. ] Building package [ linux-buster-root-current-rockpro64 ]
[ o.k. ] Building desktop package [ armbian-buster-desktop_20.11.0-trunk_all ]
[ o.k. ] Starting rootfs and image building process for [ rockpro64 buster ]
[ o.k. ] Checking for local cache [ buster-minimal-arm64.050...dfe.tar.lz4 ]
[ .... ] searching on servers
[#ac8457 0B/0B CN:1 DL:0B][ .... ] downloading using torrent network [ buster-minimal-arm64.05087badd3de0fbc857f57d1ec6badfe.tar.lz4 ]
[ o.k. ] Verified [ PGP ]
[ o.k. ] Extracting buster-minimal-arm64.050...dfe.tar.lz4 [ 0 days old ]
[ .... ] buster-minimal-arm64.050...dfe.tar.lz4:  227MiB [ 162MiB/s] [===========================================================================================================================================================================================>] 100%
[ o.k. ] Adding Armbian repository and authentication key [ /etc/apt/sources.list.d/armbian.list ]
chroot: failed to run command '/bin/bash': Exec format error
mount: /home/voltagex/src/armbian-build/.tmp/rootfs-current-rockpro64-buster-no-yes/dev: permission denied.
[ o.k. ] Applying distribution specific tweaks for [ buster ]
[ o.k. ] Applying common tweaks
chroot: failed to run command '/bin/bash': Exec format error
chroot: failed to run command '/bin/bash': Exec format error
chroot: failed to run command '/bin/bash': Exec format error
[ .... ] Updating [ package lists ]
chroot: failed to run command '/bin/bash': Exec format error
[ error ] ERROR in function install_common [ distributions.sh:216 ]
[ error ] Updating packages failed
[ o.k. ] Process terminated
[ o.k. ] Unmounting [ /home/voltagex/src/armbian-build/.tmp/rootfs-current-rockpro64-buster-no-yes/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:66 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated

```

voltagex, 2020-10-18:
Think I found the cause: 
At the start of the build -

```
mount: /proc/sys/fs/binfmt_misc: permission denied.
mount: /proc/sys/fs/binfmt_misc: permission denied.
update-binfmts: warning: Couldn't mount the binfmt_misc filesystem on /proc/sys/fs/binfmt_misc.
mount: /proc/sys/fs/binfmt_misc: permission denied.
update-binfmts: warning: Couldn't mount the binfmt_misc filesystem on /proc/sys/fs/binfmt_misc.
```

paolosabatino, 2020-10-18:
Got a similar error yesterday while I was building an rk3328 image on a couple of runs. A third run worked but produced an image with the issue https://github.com/armbian/build/issues/2260

Despite not having the "exec format error" messages, I also got the error `[ error ] ERROR in function install_common [ distributions.sh:216 ]`

My setup is ubuntu focal virtualized via kvm

igorpecovnik, 2020-10-18:
LXD container is not supported method - building images is problematic - for example, with Docker, you need to run it privilege mode: https://github.com/armbian/build/blob/master/config/templates/config-docker.conf#L32-L34

voltagex, 2020-10-19:
@igorpecovnik thanks for the info. Are there any workarounds for this? It's only the last commands that need higher privileges, right? If I'm able to invoke qemu-user I should be able to complete the build.

igorpecovnik, 2020-10-19:
> Are there any workarounds for this?

Workaround is using Docker. For LXD, sorry, don't know internals to have a clue without R&D - its certainly something with privileges.

Closing - not an issue since we don't support LXD. If you find out - you are welcome to add instructions https://docs.armbian.com/Developer-Guide_Build-Preparation/ for others that might go this way.

closed 2020-10-19 by igorpecovnik