armbian/build #2260: "Git not responding ..." strings in /etc/armbian-release

paolosabatino, 2020-10-18:
I built an armbian rk3328 image yesterday for a box of mine, now I see I have this content into /etc/armbian-release:

```
# PLEASE DO NOT EDIT THIS FILE
BOARD=rk3318-box
BOARD_NAME="RK3318 Box"
BOARDFAMILY=rockchip64
BUILD_REPOSITORY_URL=[ .... ] Git not responding. Retrying in 10 seconds. 
[ .... ] Git not responding. Retrying in 10 seconds. 
BUILD_REPOSITORY_COMMIT=52edb901e
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=20.11.0-trunk
LINUXFAMILY=rockchip64
BRANCH=legacy
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=tvb
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
```

I guess it's related to https://github.com/armbian/build/pull/2251

lanefu, 2020-10-18:
wasn't resolved by #2262

lanefu, 2020-10-18:
actually it does look like it worked... @paolosabatino can you validate?

closed 2020-10-18 by lanefu

paolosabatino, 2020-10-18:
Seems rather ok now on my side, thanks!
```
root@armbianbuild:/mnt/etc# cat armbian-release 
# PLEASE DO NOT EDIT THIS FILE
BOARD=rk3318-box
BOARD_NAME="RK3318 Box"
BOARDFAMILY=rockchip64
BUILD_REPOSITORY_URL=
BUILD_REPOSITORY_COMMIT=cb26c138d
DISTRIBUTION_CODENAME=bullseye
DISTRIBUTION_STATUS=csc
VERSION=20.11.0-trunk
LINUXFAMILY=rockchip64
BRANCH=legacy
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=tvb
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
```

PR #2262 (2020-10-18): tweak variable expansion for improved_git
`$*` expansion is impacted by `IFS` setting

switched to `$@`

should resolve quirks like https://freenode.irclog.whitequark.org/armbian/2020-10-18#28168141
 resolves #2260