armbian/build #2267: [suggestion] set RK3399 CPUMAX to 1800000 a per specs

pedrolamas, 2020-10-21:
Looking at [/config/sources/families/include/rockchip64_common.inc](https://github.com/armbian/build/blob/4492234db71c4f9646231a55b1d3551b15401e9e/config/sources/families/include/rockchip64_common.inc#L16-L19), it seems that the RK3399 is set to have CPUMAX to 2016000, but according to the [data sheet](http://opensource.rock-chips.com/images/d/d7/Rockchip_RK3399_Datasheet_V2.1-20200323.pdf) that is for the RK3399**K** not the regular RK3399.

[image]

To that fact, [this recent change](https://github.com/armbian/build/blob/4492234db71c4f9646231a55b1d3551b15401e9e/config/boards/helios64.wip#L10) was made just for the Helios64 to make it conform with the above maximum of 1800000.

I've been having issues with my NanoPi M4V2 (described [here](https://forum.armbian.com/topic/15047-nanopi-m4v2-randomly-crashes/)) and I do believe in part that they are related to the overclocking of the CPU.

So unless this was an intentional overclock of the RK3399, I would suggest that the default CPUMAX value to be set to 1800000

heiher, 2020-10-22:
Yes, I also think that overclocking caused random panic. In addition, the highest frequency of the little cores (A53) should also be set to 1.416GHz. This is also the default configuration of the official mainline kernel: 1.416GHz+1.8GHz.

Thank you.

georgefang73, 2020-10-28:
you can try to limit the max cpu speed in armbian-config, after all in some cases people want a faster cpu speed.

Cheers

jdmarshall, 2020-11-09:
@georgefang73 Shouldn't those people go set a higher speed in arbian-config, rather than leaving the rest of us with unstable systems?

(I have one that is now just rebooting in a loop after running armbian-config, but I'm not sure if it's related to armbian or a separate issue)

igorpecovnik, 2020-12-17:
Downclocked and added overlay which can be installed with armbian-config or manually.

closed 2020-12-17 by igorpecovnik