armbian/build #2268: "linux-firmware.git" Download item Mainland China address

TelDragon, 2020-10-22:
Optional parameters  

USE_MAINLINE_GOOGLE_MIRROR=""   

Only Google and source address.
Suggest to add
tuna: use mirror provided by Tsinghua University

For users in mainland China, Google is blocked

RadxaYuntian, 2021-12-08:
Did a test build with 

> ./compile.sh docker BOARD=radxa-zero BRANCH=current RELEASE=bullseye BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no DOWNLOAD_MIRROR=china **MAINLINE_MIRROR=tuna** UBOOT_MIRROR=gitee GITHUB_MIRROR=fastgit

and got working image. The last 2 options are my custom mirror options unrelated to MAINLINE_MIRROR.

qiurui144, 2022-05-23:
may be clone to gitee ,maybe fail.
lots of these proble because of this.

igorpecovnik, 2022-05-23:
Are there more reliable Github mirrors?

qiurui144, 2022-05-24:
can not be equal to github, so when i debug, also have this proble sometimes.

igorpecovnik, 2022-09-13:
Closing. Seems its an random network issue.

closed 2022-09-13 by igorpecovnik