armbian/build #2284: lib/configuration.sh changes and improved_git change problems

5kft, 2020-10-27:
Hi @igorpecovnik , the new "improved_git" retry mechanism causes a 30 second hang when I'm doing a build when "lib/configuration.sh" is executed.  I finally spent some time tracking down the problem.  It ends up that my operating on "armbian/build" as a fork  with a separate remote in the repo evidences a problem hidden a long time in "lib/configuration.sh".

This line is incorrect if the repository being used has more than one remote (in my case, I have both "origin" and "upstream" remotes):

```BUILD_REPOSITORY_URL=$(improved_git remote get-url $(improved_git remote 2>/dev/null) 2>/dev/null)```

As the inner "improved_git remote" returns two results, the outer "improved_git remote get-url" fails, and as such "improved_git" retries several times and ultimately fails...

I'm guessing that ultimately the goal here is to get the URL to the "origin" repository, so perhaps this line in "lib/configuration.sh" should be changed to:

```BUILD_REPOSITORY_URL=$(improved_git remote get-url $(improved_git remote 2>/dev/ | grep origin) 2>/dev/null)```

I.e., get the URL for "origin" for the repo?

Happy to submit a PR for this but figured I'd post as an issue so that the fix could address the proper intent of this line.

igorpecovnik, 2020-10-27:
> I'm guessing that ultimately the goal here is to get the URL to the "origin" repository, so perhaps this line in "lib/configuration.sh" should be changed to:
>
> BUILD_REPOSITORY_URL=$(improved_git [...]

For our official builds, where this matters, we only seek for origin URL. I would say this is a solution.

5kft, 2020-10-27:
> For our official builds, where this matters, we only seek for origin URL. I would say this is a solution.

OK, I can take care of this change shortly.

5kft, 2020-10-27:
Fixed with https://github.com/armbian/build/commit/709a484d226ccb0e5180f63cf555ffca29c58200; closing

closed 2020-10-27 by 5kft