armbian/build #2285: improved_git improperly retrying on real non-error conditions

5kft, 2020-10-27:
Hi @igorpecovnik , I've been suffering through builds that use the new "improved_git" changes as at random points through the build it keeps retrying for seemingly no reason.  One example is during the build of u-boot - improved_git keeps retrying for a condition that it shouldn't.  I tweaked "improved_git" to log the command executed, and to print the result.  Here's an example of the run showing the problem;

```
...
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ focal ]
[ o.k. ] Running in container [ docker ]
[ warn ] apt-cacher is disabled in containers, set NO_APT_CACHER=no to override
[ o.k. ] Checking for external GCC compilers
[ o.k. ] Downloading sources
[ o.k. ] Checking git sources [ u-boot v2020.10 ]
[ .... ] Creating local copy
/root/armbian/compile.sh: executing /usr/bin/git init -q .
/root/armbian/compile.sh: result is 0
/root/armbian/compile.sh: executing /usr/bin/git remote add origin https://github.com/RobertCNelson/u-boot
fatal: remote origin already exists.
/root/armbian/compile.sh: result is 128
/root/armbian/compile.sh: executing /usr/bin/git remote add origin https://github.com/RobertCNelson/u-boot
fatal: remote origin already exists.
/root/armbian/compile.sh: result is 128
[ .... ] Fetching updates
/root/armbian/compile.sh: executing /usr/bin/git fetch --depth 1 origin tags/v2020.10
...
```

As you can see, there is already an origin, so the command errors out and retries, causing an unnecessary long delay in the build process.

As an easy workaround, I was wondering if you might be open to the adding of a hidden build option to disable "improved_git" and just have it call "git", as that works around these retry problems.  (Personally I've never seen an issue with git and a remote that necessitated retries...so normal git is just fine.)  E.g.,

```./compile docker USE_IMPROVED_GIT=no ...```

(where USE_IMPROVED_GIT could be "yes" by default)

Just a thought.  Otherwise I can keep locally patching out "improved_git" for my builds...

igorpecovnik, 2020-10-27:
I just reverted one part of it https://github.com/armbian/build/pull/2286 since its totally not needed and it broke down patch making.

We can revert it all together but how to deal with the original problem? I have nothing against adding this variable if we don't find any better way.

5kft, 2020-10-27:
It's a good question...  From a cursory check git error codes seem to be pretty general, the severity of an error depends on the context of the git operation.  The logic of "improved_git" could be extended to retry based on the input parameters, perhaps?

igorpecovnik, 2020-10-27:
>  to retry based on the input parameters, perhaps?

Perhaps that would be a solution, yes.

5kft, 2020-10-29:
With the latest changes relating to "improved_git", everything seems to be working fine again.  Closing this issue; we can reopen if other problems appear.

closed 2020-10-29 by 5kft