armbian/build #2288: "general-packaging-..." process - patch the scripts or simply replace them with our own?

5kft, 2020-10-29:
Given that we have working "scripts/package/builddeb" and "scripts/package/mkdebian" scripts for our target build processes, does it make sense to go through the trouble of patching the upstream original in the kernel, or should we simply replace the upstream versions with our own?

The latest 5.8.17 and 5.9.2 kernels have changes that will need to be accounted for in our "patch/misc/general-packaging-..." patches, and it seems like unnecessary work to keep patching these different versions just to get the same resulting script (i.e., our own).  Of course we should keep our versions up to date if there are any significant changes made to the upstream versions that have to be included, but it seems like this shouldn't really happen that often.

igorpecovnik, 2020-10-30:
Well, we had an idea to rework this and make packaging from scratch - work already started by @The-going but he was not seen for a while.

Creating new patch is just going the easy way I guess and changes here are not that often ... I see you already made patch for 5.10.y which should probably work for 5.8.17 and 5.9.2+ ?

piter75, 2020-10-30:
> I see you already made patch for 5.10.y which should probably work for 5.8.17 and 5.9.2+ ?

IMO changes in 5.10.y are incompatible with those found in 5.8.17 and 5.9.2 so they need a different patch.

5kft, 2020-10-30:
> Creating new patch is just going the easy way I guess and changes here are not that often ... 

Yes, that's true...  I was thinking that we could get rid of this whole set of patches and simplify, but I agree what we have works and it doesn't change that often.

> I see you already made patch for 5.10.y which should probably work for 5.8.17 and 5.9.2+ ?

@piter75 is correct - the 5.10.y changes checked in are different than those for 5.8.17 and 5.9.2+.  I'm going to be keeping an eye on 5.10.y (as I need some patches that are coming in it), so I'll keep that patch up to date if these change again.

closed 2020-10-30 by 5kft

igorpecovnik, 2020-10-30:
> we have works and it doesn't change that often.

This was done: https://github.com/The-going/armbian-build/tree/packaging I have no clue how much work it represent to pull that into the operative state. 

Once we are bored I guess :)