armbian/build #2315: Wired Ethernet Fails H6 (Orange Pi3) Kernel 5.9.5, 5.10.0

lanefu, 2020-11-07:
67887860572ac4a3162fb321f3df616e430ebf75 

Wired ethernet unable to interact with network with 5.9.5, 5.10.0

https://forum.armbian.com/topic/9368-orangepi-3-h6-allwiner-chip/?do=findComment&comment=112124

@5kft would you mind taking a look?

I tried disabling `patch/kernel/sunxi-current/revert-rtl8211e-rx-tx-delay-config-change.patch` but it had no impact

5kft, 2020-11-07:
From the schematics the OPi3 uses TXDLY and RXDLY pull-ups, so the reversion patch is still needed.  Could you do an "armbianmonitor -u" and give the link to the info upload?

EvilOlaf, 2020-11-07:
FYI https://armbian.atlassian.net/browse/AR-512

5kft, 2020-11-07:
I dug up an old OPi0+ as it has the same GMAC HW configuration as the OPi3 (https://linux-sunxi.org/images/6/67/ORANGE_PI-ZERO-PLUS_V1_0_Schematic.pdf, page 10), and wired Ethernet works fine with a full ToT image build from master (5.9.5).  So it seems like it's a DT/driver configuration issue with the H6 version.  @lanefu to debug this, as a test you might try changing the H6 DT "emac" configuration (sun50i-h6-orangepi-3.dts) - e.g., leave the revert patch in place, and perhaps change the H6 DT to be as similar as possible to the OPi H3 or H5 boards.

lanefu, 2020-11-07:
Yeah I'll dig into them.  Looked main difference in dts between one plus
and 3 seems to be the name of the regulator associated with it

Tonymac32, 2020-11-07:
Mine recognizes the network interface becoming available but fails to
fully connect (dmesg says link is ready but no traffic will flow)

lanefu, 2020-11-08:
> Mine recognizes the network interface becoming available but fails to fully connect (dmesg says link is ready but no traffic will flow)

Yep that's the symptom I have as well

5kft, 2020-11-08:
> Yep that's the symptom I have as well

Yes, that's what happens when the RX/TX delays are improperly configured.  As a quick test, you could simply try leaving the revert patch in, and change the "phy-mode" in the DT to "rgmii", then see if it starts working.

EvilOlaf, 2020-11-10:
Ethernet is fine on 5.9.0-rc5 and 5.10.0-rc2 on OPi1+

gdm85, 2020-11-18:
> Mine recognizes the network interface becoming available but fails to fully connect (dmesg says link is ready but no traffic will flow)

Same here, Orange Pi 3. Last working version I had in APT cache is `linux-image-dev-sunxi64_20.05.0-trunk.133_arm64.deb`; `123` also working fine. When using `linux-image-dev-sunxi64_20.08.2_arm64.deb` the ethernet interface shows 0 RX packets, never changing. Switched to `linux-image-current-sunxi64` as this is not a box I use for experiments.

lanefu, 2020-11-19:
Disappointing update.. no success so far.   Also I noticed that the orangepioneplus dts also uses rgmii-id and that's working so confusing.   Sorry didn't have console hooked up.

have tried the following

- disable rx/tx patch
- change phy setting from rgmii-id to rgmii
- disalbe rx/tx patch and change phy setting to rgmii

lanefu, 2020-11-25:
well I removed _all_ patches from sunxi folder and built dev.

surprise it worked https://snipboard.io/Tsczwi.jpg

EvilOlaf, 2020-12-12:
> 
> 
> well I removed _all_ patches from sunxi folder and built dev.
> 
> surprise it worked https://snipboard.io/Tsczwi.jpg

What might be worth a try is compare the generated dtb files from all-patches-removed-and-working to standard build to see if there are some weird duplicates or something that breaks...

EvilOlaf, 2020-12-12:
I wonder why this patch applies against the sources without errors because half of it is already present in upstream.

https://github.com/armbian/build/blob/master/patch/kernel/sunxi-dev/board-h6-orangepi3-add-missing-eMMC.patch

https://github.com/megous/linux/blob/3f136e6829531dcbba8530ae9a61932970d37733/arch/arm64/boot/dts/allwinner/sun50i-h6-orangepi-3.dts#L205-L212

EvilOlaf, 2020-12-12:
Have you tried disabling `revert-rtl8211e-rx-tx-delay-config-change.patch`?

EvilOlaf, 2020-12-14:
https://github.com/megous/linux/commit/1b79ce32918502c9d9508e152bfa06bcbfcc7717
https://github.com/megous/linux/commit/b0842946f86bd5d0f9e857133e80d5d9496091b7
Just catched those. Need to check if they are already present.

lanefu, 2020-12-16:
@EvilOlaf  just tried... no work.... it's definitely one of our patches causing the problem.

EvilOlaf, 2020-12-16:
Also tried with disabling the patch mentioned above?

lanefu, 2020-12-16:
Ha okay it worked this time!  need to validate with other devices I guess? #2473

jernejsk, 2020-12-20:
FYI, most boards are fixed in 5.10. OPi3 doesn't have ethernet in mainline so patch for that must be adjusted. It works for me if I change phy-mode to "rgmii-txid" which is contrary to what should be, but it's possible that schematic has wrong markings... I use that setting in LibreELEC for some time now and nobody complained so far.

gdm85, 2020-12-20:
This recent update broke ethernet again (this time when using the `current` variant); excerpt from `aptitude.log`:

```
[UPGRADE] linux-bionic-root-dev-orangepi3:arm64 20.08.17 -> 20.11.3
[UPGRADE] linux-dtb-current-sunxi64:arm64 20.08.16 -> 20.11.3
[UPGRADE] linux-image-current-sunxi64:arm64 20.08.16 -> 20.11.3
[UPGRADE] linux-libc-dev:arm64 4.15.0-124.127 -> 4.15.0-128.131
[UPGRADE] linux-u-boot-orangepi3-dev:arm64 20.08.13 -> 20.11.3
```

The broken ethernet is introduced by the line `linux-image-current-sunxi64:arm64 20.08.16 -> 20.11.3`.

igorpecovnik, 2020-12-20:
> The broken ethernet is introduced by the line

Yes, this is the bug we are talking here.

lanefu, 2020-12-20:
> This recent update broke ethernet again (this time when using the `current` variant); excerpt from `aptitude.log`:
>
> ```
> [UPGRADE] linux-bionic-root-dev-orangepi3:arm64 20.08.17 -> 20.11.3
> [UPGRADE] [...]

are you running nightly?   whats in `/etc/armbian-release`

lanefu, 2020-12-20:
I have some test images available here for 5.10.y based 5ks updates to this PR  https://armbian.lane-fu.com/PR/AR-512/

gdm85, 2020-12-20:
> Yes, this is the bug we are talking here.

The bug was not in the `current` variant before, see [my previous comment](https://github.com/armbian/build/issues/2315#issuecomment-729987408).

> are you running nightly? whats in `/etc/armbian-release`

It is not nightly, but I was using the `dev` Linux kernel before Nov 18th because of an unrelated issue with the USB hub.

```
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepi3
BOARD_NAME="Orange Pi 3"
BOARDFAMILY=sun50iw6
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=b9814056
DISTRIBUTION_CODENAME=bionic
DISTRIBUTION_STATUS=supported
VERSION=20.11.3
LINUXFAMILY=sunxi64
BRANCH=dev
ARCH=arm64
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
```

lanefu, 2020-12-23:
Could use help collecting information described here

https://github.com/armbian/build/pull/2473#issuecomment-746744252

Aivaras-s, 2020-12-23:
> 
> 
> I have some test images available here for 5.10.y based 5ks updates to this PR https://armbian.lane-fu.com/PR/AR-512/

tested, network works on orangepi3

lanefu, 2020-12-30:
Fixed Merged via 7bb5101977aa6466abc3010ff0dcd4d8f41039bf

Note: Fix applied to "dev" Branch only as it is Kernel 5.10. and we are transitioning to it for next release.

Will be available in next nightly dev kernel.

closed 2020-12-30 by lanefu