armbian/build #2317: RK3328 ROCK64 won't boot via latest REPO branch

SuzukiHonoka, 2020-11-07:
I got a rock64 board, and I'd like to build the latest armbian firmware of it for enhancement.
But it always failed to boot. Even all the LEDs are not lighted.
I tried the 3 BRANCH of script itselfs's args (current,dev..) , unfortunately none of them worked.
then I compiled them via the 19.10 BRANCH.
It finally worked.
So could you guys do some revert to the uboot and the load address of rk3328 for recovery?
Thanks.

EvilOlaf, 2020-11-07:
You forked the repo earlier today. Did you do some testing to get the board boot again when build against master with adjustments?

SuzukiHonoka, 2020-11-07:
The changes I made to that forked repo has booted my devices successfully.
But I can't sure I was doing right.
I didn't know so musch about the uboot and loader so.
Thanks for the reply~

EvilOlaf, 2020-11-07:
If you found a configuration that works for you feel free to create a PR so we can review it :)

Tonymac32, 2020-11-07:
Also, can you provide the version of the Rock64 you have?  The version 2
board is not supported in any way as they are somehow defective where the
ram is concerned.

SuzukiHonoka, 2020-11-10:
> Also, can you provide the version of the Rock64 you have? The version 2 board is not supported in any way as they are somehow defective where the ram is concerned.
> […](#)

I can't assure the version of my rock64 board, I brought from OEM.
The only thing I'm sure is the mainline U-Boot doesn't work for my board.
Also the changed load address.

igorpecovnik, 2020-11-10:
This is build script bug tracker and from this perspective there is nothing wrong.

Try to use [community forums](https://forum.armbian.com/forum/23-community-forums/) for best effort research and support. Armbian had officially dropped support for this board due to quality reasons and related high support costs.

https://armbian.atlassian.net/browse/AR-520

closed 2020-11-10 by igorpecovnik