armbian/build #2349: linux-dtb-current-sunxi64.postinst: Exec format error   ---   Missing shebang

FredericGuilbault, 2020-11-15:
Hi, I have a big bash script that download armbian and mount the iso, chroot into it and  generate a flavoured version of armbian. (the script is doing the same with raspbian, and debian)

First thing the script do is `apt update && apt upgrade`   
Since I upgraded the version if armbian Im downloading to  `Armbian_20.05.2_Pine64so_buster_current_5.4.43.img` the following error have appeared :
 
```
Preparing to unpack .../06-linux-dtb-current-sunxi64_20.08_arm64.deb ...
Error while loading /var/lib/dpkg/tmp.ci/preinst: Exec format error
dpkg: error processing archive /tmp/apt-dpkg-install-xF9W1s/06-linux-dtb-current-sunxi64_20.08_arm64.deb (--unpack):
 new linux-dtb-current-sunxi64 package pre-installation script subprocess returned error exit status 1
Error while loading /var/lib/dpkg/info/linux-dtb-current-sunxi64.postinst: Exec format error
dpkg: error while cleaning up:
 installed linux-dtb-current-sunxi64 package post-installation script subprocess returned error exit status 1
```

I googled a bit and found this post :
https://askubuntu.com/questions/1001795/unable-to-execute-new-pre-installation-script-var-lib-dpkg-tmp-ci-preinst

It stipulates that (pre|post)inst files in the package should start with a shebang, I have checked and it's  effectively missing in the package. `06-linux-dtb-current-sunxi64_20.08_arm64.deb`

Note that running the `perinst` file straight on the prompt does not trow error. It happens only when doing `apt update && apt upgrade`  

--Fred

EvilOlaf, 2020-11-15:
The official supported way to build Armbian is using the build script as it is. We cannot provide support for any 3rd party modifications.
Also user customization while building is possible: https://docs.armbian.com/Developer-Guide_User-Configurations/

lanefu, 2020-11-15:
@FredericGuilbault that bin exec error makes me think that you are not executing this on an ARM64 device.  What CPU architecture are you chrooting on

FredericGuilbault, 2020-11-16:
@EvilOlaf no problems, im just reporting a missing shebang in the package linux-dtb-current-sunxi64

From a policy perspective, debian is clear about the need of a shebang in the preinst/postinst. 
https://www.debian.org/doc/debian-policy/ch-maintainerscripts.html

and as proof of concept, doing this fix the package (for me):
```
apt download linux-dtb-current-sunxi64
dpkg-deb -R linux-dtb-current-sunxi64_20.08.16_arm64.deb tmp
sed -i '1s;^;#!/bin/bash\n;' ./tmp/DEBIAN/postinst
sed -i '1s;^;#!/bin/bash\n;' ./tmp/DEBIAN/preinst
dpkg-deb -b tmp fixed.deb
dpkg -i fixed.deb
```

I searched the present repository for this specific package and I didn't find it :(

If I have confirmation that's something you want to fix and know where are the DEBIAN files of this package. I can make a PR.   

___

@lanefu 
Im using `proot` to run it with qemu-aarch64.  

`Armbian_20.02.0-rc1_Pine64so_buster_current_5.4.14.img` was working 
`Armbian_20.05.2_Pine64so_buster_current_5.4.43.img`  is not working

igorpecovnik, 2020-11-16:
> I searched the present repository for this specific package and I didn't find it :(

location: https://github.com/armbian/build/tree/master/patch/misc
files: general-packaging-*

I guess all packaging patches needs attention.

FredericGuilbault, 2020-11-18:
oh, ok....      I was naively expecting a small folder with debianization files.  I guess I will have to read your doc about how all of this work :) . 
Thx

igorpecovnik, 2020-11-22:
> I was naively expecting a single folder with debianization files

:) 

Debian / Ubuntu is just a compatibility layer. We use a selection of their prebuilt userland packages and retain package relations (we also usually don't fix bugs on this level unless its critical) to achieve that. Practically everything else and especially hardware interface, the expensive part of the equation, is our work.

FredericGuilbault, 2020-11-27:
ok. So If I get it right,
Inside the file `/patch/misc/general-packaging-5.10.y.patch`
I find my target at 
```
+++ b/scripts/package/builddeb
~~~
+	cat >> $pdir/DEBIAN/preinst <<EOT
+rm -rf /boot/dtb-$version; rm -rf /boot/dtb
+exit 0
+EOT
```
This is where the shebang is missing but the file `scripts/package/builddeb` don't exist in the repo

After trying every option in the build script trying to download most source files as I can. 

The only occurrence of what the file should look like after it been  patched is : 
`/cache/sources/linux-mainline/orange-pi-5.9/scripts/package/builddeb `

But im confused cuz it's `orange-pi` not `Pine64so`  

Do they relate, am I at the good place ?

lanefu, 2020-11-28:
> But im confused cuz it's orange-pi not Pine64so
> Do they relate, am I at the good place ?

Yep that's correct... Both are sunxi (Allwinner SoCs). We use megous Orange Pi kernel repo for Allwinner SoCs

FredericGuilbault, 2020-12-05:
IDK how but my original problem have been solved in the last release. Armbian update seamlessly in my usecase now. 

Do you still want to correct this or you prefer to close the issue ?

igorpecovnik, 2020-12-05:
> Do you still want to correct this

We do - but didn't find time.

FredericGuilbault, 2020-12-05:
Ill keep on my list too then

rpardini, 2023-06-01:
Let's close this. We've bash shebangs for all kernel postinsts and similars now. https://github.com/armbian/build/blob/main/lib/functions/compilation/kernel-debs.sh#L153

closed 2023-06-01 by rpardini