armbian/build #2351: [request] change CONFIG_DRM_LIMA and CONFIG_DRM_PANFROST to "y" for linux-rockchip-dev.config

q4a, 2020-11-16:
Hi. Please, set
```
CONFIG_DRM_LIMA=y
CONFIG_DRM_PANFROST=y
```

instead of current:
```
CONFIG_DRM_LIMA=m
CONFIG_DRM_PANFROST=m
```
at least for [linux-rockchip-dev.config](https://github.com/armbian/build/blob/b8af894293134fc471003d2f0c2e0a87d10dddf6/config/kernel/linux-rockchip-dev.config#L4304) and may be some other configs like [linux-rockchip-current.config](https://github.com/armbian/build/blob/b8af894293134fc471003d2f0c2e0a87d10dddf6/config/kernel/linux-rockchip-current.config#L4316) at your discretion.
It is already done in the [linux-odroidxu4-dev.config](https://github.com/armbian/build/blob/b8af894293134fc471003d2f0c2e0a87d10dddf6/config/kernel/linux-odroidxu4-dev.config#L4216)

Thanks.

igorpecovnik, 2020-11-16:
> I think, that many people would like to test panfrost driver for their boards.

It should work as a module just fine. Doesn't it?

EvilOlaf, 2020-11-16:
`modprobe panfrost`

Tonymac32, 2020-11-16:
Panfrost loads by default on my RK3399 and RK3288 boards as a module.

These config files are for the Tinker Board and MiQi, just so everyone is
clear, Lima should not be enabled as the RK3288 is the only SoC supported.

q4a, 2020-11-16:
Sorry, I fixed error with my setup and Panfrost works with current config.

closed 2020-11-16 by q4a