armbian/build #2356: NanoPi Neo3 not reconized

Kebims, 2020-11-17:
Hi,

NanoPi Neo3 is reconized as NanoPi R2S.

I think there is a problem in Device-Tree because when I start my board with debug log, I receive this output

`U-Boot 2020.04-armbian (Jul 14 2020 - 20:01:58 +0000)

Model: FriendlyElec NanoPi R2S
DRAM:  2 GiB
PMIC:  RK8050 (on=0x40, off=0x00)
MMC:   rksdmmc@ff500000: 1
Loading Environment from MMC... *** Warning - bad CRC, using default environment`

I found source code from riptidewave93 : [https://github.com/riptidewave93/rk3328-debian](url) but I don't know how to patch device-tree.

Thanks.

igorpecovnik, 2020-11-17:
Boards that doesn't differ much can share boot loader. The only difference is kernel DT specified here:
https://github.com/armbian/build/blob/master/config/boards/nanopineo3.conf#L11
Everything is fine.

Kebims, 2020-11-17:
Sure but there is a difference with hardware MAC address on I²C bus ( 24AA025E48 ).

igorpecovnik, 2020-11-17:
You can try adding dedicated DT file for U-boot similar to:
https://github.com/armbian/build/blob/master/patch/u-boot/u-boot-rockchip64-mainline/add-board-nanopi-r2s.patch

Kebims, 2020-11-20:
Okay, I'll try.
As this is the first time I'm creating a patch, so it may take me a long time to do it.

EvilOlaf, 2021-03-26:
Assuming solved.

closed 2021-03-26 by EvilOlaf