armbian/build #2379: Ubuntu Focal should purge package systemd-timesyncd

PanderMusubi, 2020-11-25:
On Ubuntu Focal 20.08.17 and 20.11.0-trunk.41 nightly (on Pine A64), the package systemd-timesyncd is removed but not completely purged, see

    dpkg -l systemd-timesyncd

which returns `rc`. This can be solved with

    dpkg -P systemd-timesyncd

This issue is only from a housekeeping perspective. Solving this will prevent users wondering if that package should be reinstalled or purged or if something is amiss.

PanderMusubi, 2021-02-15:
This issue is also on Ubuntu Groovy.

EvilOlaf, 2021-02-15:
I could not find any relation to `systemd-timesyncd` within the build script. Therefore I assume Armbian does not touch this package and it is upstream issue.

lanefu, 2021-02-15:
It could be a result of Armbian shipping with `chrony` for NTP...

EvilOlaf, 2021-02-15:
Good catch. `chrony` removes `systemd-timesyncd` when being installed automatically.

PanderMusubi, 2021-02-17:
Indeed, installation of chrony gives systemd-timesyncd an `rc` status in `dpkg -l` which is solved with the `dpkg -P` from above.

EvilOlaf, 2021-02-17:
Not sure if the `--purge` switch is compatible with `install` but you can try to add it here and build an image to see what happens: 
https://github.com/armbian/build/blob/705bbd431bd2f3751c74a8a84eb4d4a4cb3a09e8/lib/debootstrap.sh#L235-L239

PanderMusubi, 2021-02-17:
Not that I know. How about adding a step after this one that purges all removed packages? Will also catch similar cases in teh future.

EvilOlaf, 2021-02-17:
Well, the `4.need PR` label is there ...😄

PanderMusubi, 2021-02-17:
Will make one! Just wanted to check that the direction of solving this is supported.

EvilOlaf, 2021-02-17:
I cannot guarantee that it would be accepted. My knowledge in this part is not deep enough to make that decision also thinking about possible side effects.
Also worth mentioning that this is actually a very minor, almost cosmetical, issue.

lanefu, 2021-02-18:
Pr looks sane I’ll try to test it soon

closed 2021-02-20 by PanderMusubi