armbian/build #2392: Docker build uses unsupported Ubuntu Bionic

cybe, 2020-11-29:
Hey,

with #2316 bionic was marked as unsupported as a host build environment. However, it is still being used while [Building with Docker](https://docs.armbian.com/Developer-Guide_Building-with-Docker/):

```
./compile.sh docker
[ o.k. ] Using config file [ /var/lib/docker/dennis/dev/armbian-build/userpatches/config-docker.conf ]
[ .... ] Using existed a armbian Docker container
[ o.k. ] Running the container
[ o.k. ] Using config file [ /root/armbian/userpatches/config-odroidn2.conf ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ bionic ]
[ error ] ERROR in function prepare_host [ general.sh:1003 ]
[ error ] It seems you ignore documentation and run an unsupported build system: bionic
```

EvilOlaf, 2020-11-29:
Docs seem outdated

cybe, 2020-11-29:
My actual host OS is focal:
```
$ lsb_release -d
Description:    Ubuntu 20.04.1 LTS
```
Bionic is somewhere referenced during the Docker build itself.

igorpecovnik, 2020-11-29:
I think you need to start from scratch - purge old Docker images.

cybe, 2020-11-29:
> I think you need to start from scratch - purge old Docker images.

Tried that already (`docker system prune -a`). Also removed the local volumes `armbian-cache` and `armbian-ccache`.

cybe, 2020-11-29:
> Is that Dockerfile not supposed to be there?

Replying to myself: Indeed, files in userpatches won't be updated. ([reference](https://github.com/armbian/build/blob/58740edbd8dfbb947aa5cb2aec44c1d764f2ae10/compile.sh#L208)).

Sorry for the noise!

closed 2020-11-29 by cybe

EvilOlaf, 2020-11-29:
Docs received a fix. Worth it 😀